GlobalVent/wiki #5: Pick a rpi software version

ralphBellofattoSTX, 2020-04-05:
We need to standardize the rpi software version we are going to put our software on top of.

This issue is to:
* pick the version (probably buster).
* install it and try it.
* document what other stuff I needed to do to make it usable.
* find out where long term copies of this software are
  * or make a copy of it on our google drive.

ralphBellofattoSTX, 2020-04-05:
Current Raspbian version is buster, kernel 4.19
https://www.raspberrypi.org/downloads/raspbian/
There are three flavors.

1. Raspbian Buster with desktop and recommended software: https://downloads.raspberrypi.org/raspbian_full_latest
2. Raspbian Buster with desktop: https://downloads.raspberrypi.org/raspbian_latest
3. Raspbian Buster Lite: https://downloads.raspberrypi.org/raspbian_lite_latest

Starting with option 1.

ralphBellofattoSTX, 2020-04-05:
Raspbian installation guide: downloaded:
https://downloads.raspberrypi.org/raspbian_full/images/raspbian_full-2020-02-14/2020-02-13-raspbian-buster-full.zip

then follow this
https://www.raspberrypi.org/documentation/installation/installing-images/README.md
* unzip the image downloaded above (getting the 2020-02-13-raspbian-buster-full.img file)
The site recommends using the raspberry Pi imager:
https://www.raspberrypi.org/downloads/

Downloading: https://downloads.raspberrypi.org/imager/imager.dmg

Make absolutely SURE there are not other SD cards in the system.

and installing.
* choose os:
[image]

* use custom:
[image]

* locate the image we just downloaded above.

* choose SD

And.... hm... it does not work....
sd card shows nothing on the screen...
[image]

ralphBellofattoSTX, 2020-04-05:
when done, connect up an hdmi monitor and usb keyboard.
Will need both to get the initial network configuration setup.

either connect the computer to the local network or configure the wireless.

Then configure the ssh connection for the rpi user id.

Remove the SD card from the adapter. 
insert into the rpi with an hdmi monitor and usb keyboard attached and plug it in.

ralphBellofattoSTX, 2020-04-05:
NOTE: this configuration has way too much software in it for use as a basic controller, 
This configuration is the first expermental configuration to see if we can actually do software development on this system.

next I'm going to setup SSH and VNC for remote access..

[image]

[image]

[image]

[image]

[image]

[image]

[image]
[image]
[image]

ralphBellofattoSTX, 2020-04-06:
after reboot, you should be able to both ssh AND use vnc with the system.

since we are now headless (no hdmi connection, the screen resoultion is a bit small).

open a terminal or ssh session:
pi@rpi01:/boot $ sudo raspi-config 

[image]

[image]

[image]

reboot:
pi@rpi01:/boot $ sudo reboot

ralphBellofattoSTX, 2020-04-06:
NOTE: when testing the software interface without the tft display we can play with this resolution to be able to approximate what the tft display size is.

TechnologyClassroom, 2020-04-07:
I would suggest Raspbian Desktop for this.  It is a smaller download and leaves out a bunch of proprietary cruft that you should not be using for this project anyway.

ralphBellofattoSTX, 2020-04-07:
Please point to the exact version on the rpi site you would like to suggest here...

ralphBellofattoSTX, 2020-04-07:
I can repeat the process and then. when we are satisfied, i'll create a wiki page outlining a "standardized development env"

ralphBellofattoSTX, 2020-04-07:
@TechnologyClassroom 
So you are suggesting a debian release, rather than raspbain.

However, when i look for this, i see: 
https://www.raspberrypi.org/downloads/raspberry-pi-desktop/

which says:

> Raspberry Pi Desktop
> Debian with Raspberry Pi Desktop is the Foundation’s operating system for PC and Mac. You can create a live disc, run it in a virtual machine, or even install it on your computer.
> 
> Raspberry Pi Desktop comes pre-installed with plenty of software for education, programming and general use; including Python, Scratch, Sonic Pi, Java, and more.

So..... this does not look like it runs on the raspberry pi, but is an intel-based release to be run on a virtual machine.

are you suggesting the raspberry pi desktop to explore the UI?  in a VM,  or an x86 machine.  and then compile natively or use a cross compiler to build the rpi images?

TechnologyClassroom, 2020-04-07:
@ralphBellofattoSTX I was suggesting the second one you listed above from https://downloads.raspberrypi.org/raspbian/images/ specifically https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/2020-02-13-raspbian-buster.zip

ralphBellofattoSTX, 2020-04-07:
Ok, i agree.
This will eventually go on a setup wiki page.
I'm going to install it and see if there is additional software needed to make it operational.

TechnologyClassroom, 2020-04-07:
The main difference is that it removes nonfree binary software that companies allow Raspbian to distribute.  Depending on any of that software would make the workflow incompatible with other single board computers.  This is important to reduce the dependence on the Raspberry Pi supply chain.

ralphBellofattoSTX, 2020-04-07:
This selection test is about getting us something we can write software against, it may or may not be what we finally use.

Should we be using raspbain? Or one of the alternate distros.  My original feeling was that raspbain has the most device hours on it, so would likely to be more stable, but that is not based on any experience. 

since we have multiple people favoring QT as the graphics framework for displaying on the rPI i also wanted the most rapid method to figure out what we need to do to get the UI portion up and get a quick proof of concept of the tool chain needed to produce that sofware.

Also get an idea on compile and link times on the rpi nativly, vs trying to setup a cross compiler environment in a vm (possibly vagrent)...

TechnologyClassroom, 2020-04-07:
For the Raspberry Pi, I always recommend Raspbian.  There are other distributions, but the Rasbian is the most stable release.

Compile times are really slow.  I would recommend installing binaries from the standard repository instead of compiling yourself.  If you have to compile, look into cross compilation.

ralphBellofattoSTX, 2020-04-07:
the issue here is how do we do development in C++ (assuming that is the language we end up choosing) using the QT framework.

what is needed is a proof of concept for qt on the raspberry pi.
either compiling natively or in a cross compiler environent, preferably in a vagrent.

In addition to a QT compiler, we would need to demonstrate that we have a viable path forward:

See this list of tasks.
https://github.com/JamVent/wiki/wiki/initial-workbreakdown

I'm thinking it is time to two tasks (github issues) for 
* a proof of concept for C++/QT on the raspberry pi
  * native compile
  * cross compile

Starting with the native compile.

TechnologyClassroom, 2020-04-07:
I see python listed in the language list.  That would probably be best as you do not have to compile or cross compile.

ralphBellofattoSTX, 2020-04-07:
I think this discussion deserves its own github issue, i'll go create it and add you as an interested party.

https://github.com/JamVent/wiki/issues/10

closed 2020-04-14 by ralphBellofattoSTX