GlobalVent/wiki #12: Acquiring/Making Test Lung

achortos, 2020-04-07:


achortos, 2020-04-08:
Kbpeng posted this in the slack yesterday:

https://docs.google.com/document/d/1evt3kwgM_CXjX5V-56j9tzBxAO8izMrYzBY0-v9zEBo/edit

achortos, 2020-04-08:
Are there any preferences on types of test lung? If not, I could start contacting these sources to see if they have stock.

JMathiszig-Lee, 2020-04-09:
We've Managed to get our hands on a test lung in the UK that should be arriving imminently

We had been using a 2l intersurgical bag with lengths of tubing for resistances however the flow through narrow tubing was very turbulent and we realised the bag was actually lower compliance then we had realised.

carlyjb, 2020-06-08:
Needs some planks and rubberbands to set up, Joost to CAD a thing for 3d printing

carlyjb, 2020-06-15:
Current set up is good enough for our use

closed 2020-06-15 by carlyjb