GlobalVent/wiki #15: QT proof of concept prototype

ralphBellofattoSTX, 2020-04-07:
QT proof of concept prototype (for those non software types, this is a software  GUI framework that some people on this project have experience with).

* experiment with the setup a QT environment to either compile nativily on the rpi, or as across compiler.  
* document a working setup and describe how to get it setup.
* create a demo project showing the interface that has been proposed.
* demonstrate the USE of QT designer for the above demo interface.
* describe to use QT designer
* produce a top level Makefile that when the directions here are followed, "just works"

stephmusinsky, 2020-04-12:
Set up Vagrant Ubuntu vm (vagrant init hashicorp/bionic64) and then vagrant up.
http://wapel.de/?p=641
Following this with some notes. Instead of using git, downloading all qt:
wget http://download.qt.io/official_releases/qt/5.12/5.12.5/single/qt-everywhere-src-5.12.5.tar.xz
tar-xvf qt-everywhere-src-5.12.5.tar.xz
Then same ./configure line

I had to do needed to do sudo apt-get update and sudo apt-get g++

stephmusinsky, 2020-04-12:
Also installed these libraries (step 5) https://wiki.qt.io/RaspberryPi2EGLFS

ralphBellofattoSTX, 2020-04-13:
@stephmusinsky found this this set of directions to work the best.

@ralphBellofattoSTX will be taking over the rest of the QT demo and demonstrate this on a rpi TFT display.

Expect more copious notes about this adventure.

https://github.com/JamVent/wiki/wiki/VisualGDB-Setup-for-Raspberry-Pi-Development

This will eventually require a licensed copy of VisualGDB...  If that does not pan out, then we need to get a plan B.

ralphBellofattoSTX, 2020-04-13:
Setting up a windows VM on my mac to try to set this up as per the directions here:

https://github.com/JamVent/wiki/wiki/VisualGDB-Setup-for-Raspberry-Pi-Development

did the following
* installed windows vm with parallels for mac.
* downloaded the free Visual Studio Community Edition from here: https://visualgdb.com/vscommunity
* installed the "try beta" for VisualGDB here: https://visualgdb.com/download/
* TBD...

ralphBellofattoSTX, 2020-04-14:
in setting up the visual studio, it says that visualC++ is not installed and provides a link to how to install it.

[image]

interesting since this is a cross compile thing?

ralphBellofattoSTX, 2020-04-14:
try to install this:
[image]

ralphBellofattoSTX, 2020-04-14:
The directions say to pick msbuild.

[image]

however, since this is a QT proof of concept. select qt, and a remote build.

[image]

ralphBellofattoSTX, 2020-04-14:
and we got this little diddy:
[image]

ralphBellofattoSTX, 2020-04-14:
and we got:
[image]

[image]

and got this:
[image]

so... install cmake on the rpi...

also the TFT display winked out... don't know why...

ralphBellofattoSTX, 2020-04-14:
ssh into the device (rpi01.local) and did
sudo apt-get install cmake

ralphBellofattoSTX, 2020-04-14:
backtrack, and select the cross compiler option.

ralphBellofattoSTX, 2020-04-14:
and we got this:
[image]

ralphBellofattoSTX, 2020-04-14:
backtracking and using the MSBuild we get:
[image]

say yes to install it.

ralphBellofattoSTX, 2020-04-14:
hand copied the demo over... and it then worked:
```
pi@rpi02:~ $ chmod +x visualGdbDemo 
pi@rpi02:~ $ ./visualGdbDemo 
Hello, World!
```

ralphBellofattoSTX, 2020-04-14:
trying this combination
https://visualgdb.com/tutorials/linux/qt/cmake/

ralphBellofattoSTX, 2020-04-14:
got this result:
```
Running CMake: C:\Users\ralphbellofatto\AppData\Local\VisualGDB\CMake\bin\cmake.exe ../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DTOOLCHAIN_ROOT=C:/SysGCC/raspberry -DCMAKE_TOOLCHAIN_FILE=X:/github/prototypes/qtStarterApp/toolchain.cmake -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: C:/SysGCC/raspberry/bin/arm-linux-gnueabihf-gcc.exe
-- Check for working C compiler: C:/SysGCC/raspberry/bin/arm-linux-gnueabihf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/SysGCC/raspberry/bin/arm-linux-gnueabihf-g++.exe
-- Check for working CXX compiler: C:/SysGCC/raspberry/bin/arm-linux-gnueabihf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
CMake Error: AUTOMOC for target qtStarterApp: Test run of "moc" executable "C:/SysGCC/raspberry/Qt/v5-CMake/moc.exe" failed.
C:/SysGCC/raspberry/Qt/v5-CMake/moc.exe -h

Exit code 0xc0000135

CMake Generate step failed.  Build files cannot be regenerated correctly.
System.Exception: CMake exited with code 1
   at hn2.o(g a)

========== Project Configuration Summary ==========
    qtStarterApp  configured in 00:40
========== Configuration: 1 Succeeded, 0 Failed ==========
```

and
[image]

```
Loaded X:\github\prototypes\qtStarterApp\qtStarterApp.vgdbcmake. Waiting for VS to set the active configuration...
X:\github\prototypes\qtStarterApp\qtStarterApp.vgdbcmake: active configuration changed to Debug
```

ralphBellofattoSTX, 2020-04-14:
and rebuild does:
[image]

```
Removing X:\github\prototypes\qtStarterApp/VisualGDB/Debug...
cmake ../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DTOOLCHAIN_ROOT=C:/SysGCC/raspberry -DCMAKE_TOOLCHAIN_FILE=X:/github/prototypes/qtStarterApp/toolchain.cmake -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
-------------------------------------------------------------
Command exited with code -1
Executable: cmake
Arguments: ../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DTOOLCHAIN_ROOT=C:/SysGCC/raspberry -DCMAKE_TOOLCHAIN_FILE=X:/github/prototypes/qtStarterApp/toolchain.cmake -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
Directory: X:\github\prototypes\qtStarterApp/VisualGDB/Debug
Command-line action failed

========== Project Rebuild Summary ==========
    qtStarterApp  rebuilt in 00:00
========== Rebuild: 0 Succeeded, 1 Failed ==========
```
So, what is it complaining about.

ralphBellofattoSTX, 2020-04-14:
looking at this section:
https://github.com/JamVent/wiki/wiki/VisualGDB-Setup-for-Raspberry-Pi-Development#qtbuild

> Qtbuild
> Now follow the instructions here to create and build a Qt demo app. NOTE: the one deviation that I had to make was to explicitly change the Pi's path to CMake binary from VisualGDB/$(ConfigurationName) to /usr/local/bin:
> 

There is a note about the cmake binary.  so where do we find that...

ralphBellofattoSTX, 2020-04-14:
trying the build on the pi now, to see if that works any better....

making the remote compiled app and we get:
```
Running CMake: cmake ../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF
CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/home/pi/qtStarterApp/VisualGDB/Debug/CMakeFiles/CMakeOutput.log".
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
System.Exception: CMake exited with code 1
   at hn2.o(g a)

========== Project Configuration Summary ==========
    qtStarterApp  configured in 00:00
========== Configuration: 1 Succeeded, 0 Failed ==========
```

ralphBellofattoSTX, 2020-04-14:
should we use gnu make instead?

stephmusinsky, 2020-04-14:
5.12.5 QT installed on windows- was told 5.12 is most reliable

stephmusinsky, 2020-04-14:
https://www.qt.io/download-open-source

ralphBellofattoSTX, 2020-04-15:
Progress, getting the qt install and at a matched version helped here, along with directions to get it installed on windows.

TODO: add this to the wiki page

Working with the cross files, and a qt starter app. 
[image]

And the initial  file generation:
```
Running CMake: C:\Users\ralphbellofatto\AppData\Local\VisualGDB\CMake\bin\cmake.exe ../.. -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DTOOLCHAIN_ROOT=C:/SysGCC/raspberry -DCMAKE_TOOLCHAIN_FILE=X:/github/prototypes/qtStarterAppCross/toolchain.cmake -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
-- Configuring done
-- Generating done
-- Build files have been written to: X:/github/prototypes/qtStarterAppCross/VisualGDB/Debug

========== Project Configuration Summary ==========
    qtStarterAppCross  configured in 00:00
========== Configuration: 1 Succeeded, 0 Failed ==========
```

ralphBellofattoSTX, 2020-04-15:
attempting a build next...
Actions --> BuildSolution
```
C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe 
[1/5] Automatic MOC and UIC for target qtStarterAppCross
[2/5] Building CXX object CMakeFiles/qtStarterAppCross.dir/qtStarterAppCross_autogen/mocs_compilation.cpp.o
[3/5] Building CXX object CMakeFiles/qtStarterAppCross.dir/MainWindow.cpp.o
[4/5] Building CXX object CMakeFiles/qtStarterAppCross.dir/qtStarterAppCross.cpp.o
[5/5] Linking CXX executable qtStarterAppCross

========== Project Build Summary ==========
    qtStarterAppCross  built in 00:05
========== Build: 1 Succeeded, 0 Failed ==========
```

but it is not showing up on the target
So at what point does it transfer the exeutable to rpi01.local:/home/pi

ralphBellofattoSTX, 2020-04-15:
Debug --> Quick Debug with GDB

Resulted in an xwindows display going up on the windows session, not on the display...

but it did load the program on the device.
it now showed up on the device:
[image]

and running it we got:
[image]

ralphBellofattoSTX, 2020-04-15:
NotaBene: when running the app from the Visual studio, at least with the default settings, it displays the app in an Xwindow.
on the local machine, rather than on the device itself.

So, how do we get the display to go to the device itself?

ralphBellofattoSTX, 2020-04-15:
next thing to figure is how to setup the .gitignore such that only those files that are needed by the build go in.

This environment probably means that even our command line compiles have to work from windows unless cmake is smart enough..

ralphBellofattoSTX, 2020-04-15:
trying to edit the UI file now:
double click here:
[image]
and we get:
[image]

however the only qt stuff it can find is:
[image]

ralphBellofattoSTX, 2020-04-15:
Thanks Stephanie:

Here is where qt designer lives.

[image]

ralphBellofattoSTX, 2020-04-15:
adding these files to the top level .gitignore for the entire prototypes directory:
```
# visual Gdb
CodeDB/
VisualGDB/
VisualGDBCache/
*.vgdbcmake.*.user
CMakeLists.txt.old
CMakeFiles/
CMakeCache.txt
```

to only have under source control what is needed

ralphBellofattoSTX, 2020-04-15:
don't quite have a way to run this from the command line yet, but at least we can do it interactively...

the web says one needs the build tools for visual studio for command line operations:
https://docs.microsoft.com/en-us/cpp/build/building-on-the-command-line?view=vs-2019#how-to-use-the-command-line-tools

and here are the tools:
https://visualstudio.microsoft.com/downloads/#build-tools-for-visual-studio-2017

ralphBellofattoSTX, 2020-04-15:
here is something curious the "deploy. button" is not enabled
[image]

and clicking on it does nothing...

ralphBellofattoSTX, 2020-04-15:
starting the app caused it to deploy, but the xserver is not working...

[image]

not that we really want it to do an server, i would rather it run the app on its local display...

ralphBellofattoSTX, 2020-04-16:
I really don't want an xserver, but want the app to display on the device's screen or vnc session, so.... how do we do that...

ralphBellofattoSTX, 2020-04-16:
put this at the very last line in the /home/pi/.bashrc file
```
export DISPLAY=:0
```

and now the x11 app goes directly to terminal and VNC session if headless.

ralphBellofattoSTX, 2020-04-16:
asked the question of VisualGDB support and got this:

> Hi,
> 
> Please try using the VisualGDB Project Properties -> Debug Settings -> Linux GUI (X11) setting (bottom of the page).
> Choosing "Show on Target" will set the DISPLAY variable to ":0". 
> You can also keep the setting unchanged and override the environment via Debug Settings -> Common Settings -> Additional Environment.
> 
> If you would like to troubleshoot X11 forwarding, please try using SmarTTY. It uses the same SSH engine as VisualGDB, so you can experiment with setting various DISPLAY values and running the program manually. Also disabling your firewall/antivirus might help, as sometimes it would interfere with X11 connections between VisualGDB and XMing. 
> 
> --
> Best regards,
>   Sysprogs Support 
>   View conversation thread online
>

ralphBellofattoSTX, 2020-04-17:
Got this answer from the VisualGDB guys...

> Hi,
>
> Please try using the VisualGDB Project Properties -> Debug Settings -> Linux GUI (X11) setting (bottom of the page).
> Choosing "Show on Target" will set the DISPLAY variable to ":0".
> You can [...]

So, the DISPLAY=:0 gets us the results we wanted.

ralphBellofattoSTX, 2020-04-17:
Also, the topic of QML came up and we got this answer from the VisualGDB guys.

> Hi,
> 
> Sorry, VisualGDB only supports the C/C++ debugging based on the GDB debugger. It does not support JavaScript/QML debugging. 
> 
> --
> Best regards,
>   Sysprogs Support 
>   View conversation thread online
>

jwildhome, 2020-08-11:
I think we can consider this closed.  We have a Qt prototype GUI in
 https://github.com/JamVent/prototypes.git  jamventProto/jamventgui/

And we are using VisualGDB with MS Visual Studio Cmake projects as our development cross-compilation environment.

closed 2020-08-11 by carlyjb