GlobalVent/wiki #16: Proposed software architecture.

ralphBellofattoSTX, 2020-04-07:
Here is a proposed architecture.
* separate the UI from the underlying control section either with threads or, separate processes.
* separate managers for the valve manager and sensor manager to replacement with a simulator module.

[image]

ralphBellofattoSTX, 2020-04-07:
Software setup with simulator:
[image]

samcervantes, 2020-04-07:
Our remote monitoring/aggregation dashboard at www.ventilatormonitor.org (web-browser / Qt desktop app) needs an API endpoint in order to connect to each ventilator. We've written a python module to do this on a raspberry pi (which runs a flask server). With a little bit of work, this module can talk to your Control Manager layer, with the result of exposing a REST API endpoint for each Jamvent ventilator. All of the Jamvent ventilators on the network can then be polled from the same central remote interface.

ralphBellofattoSTX, 2020-04-08:
@samcervantes I'm assuming that the device exposes a rest api.  Hopefully we can just find an open source c++ rest implementation.  The timeframe to implement what we need is iffy, even without the effort, but i'll put it into the architecture doc.

[image]

[image]

samcervantes, 2020-04-09:
Oh - my mistake - I thought you were using python a the raspberry pi. If you're using C++ then the problem is simplified. We already have a C++ serial library written which can run on your raspberry pi and interface with our remote monitoring application. We're developing the serial library to be as minimal an un-obtrusive as possible. If you let me know when you have a firmware MVP then our developer can test the library on it.

ralphBellofattoSTX, 2020-04-09:
@samcervantes we are actually leaning to using C++ and QT,  python may still be selected, if we run into serious problems with this approach...

A c++ serial library could work.  We would need to work out how to share code revisions with this library in a controlled manner.

samcervantes, 2020-04-11:
@ralphBellofattoSTX - sure that makes sense. Right now we have a repo for our arduino application at https://github.com/Open-Ventilator-Remote-Monitoring/ventilator-monitor-arduino. This repo contains our serial library in addition to our demo application (which simulates and arduino and generates random sample data). It sounds like it would be best for us to split out the arduino library into a separate module so that we can control that separately. I'll ask the team to do that.

ralphBellofattoSTX, 2020-04-11:
what exactly does this library when running on the arduino?

samcervantes, 2020-04-11:
The C++ serial library simply sends ventilator data to over the USB serial port. So to integrate this with your ventilator, you simply include the library and call the function at regular intervals, passing it the data you'd like to send. Here's an example usage:

```omUpdateVentilatorData(ieRatio, peakInspiratoryPressure, tidalVolume, respiratoryRate, peep)```

You will have to decide when & how often this function is called. We recommend every ~3 seconds. 
On the other end of that USB serial connection (close by) is a separate raspberry pi, which runs a flask webserver. This new raspberry pi becomes a network endpoint for our central monitoring application. The central monitoring application then polls each ventilator endpoint at periodic intervals and displays all of the ventilator data on one convenient dashboard. So in this specific instance we would have the Jamvent raspberry pi (ventilator control) connected our separate raspberry pi (API endpoint). We designed the API endpoint application to run on a raspberry pi because most emergency ventilators run on Arduino's and do not have ethernet/wiki. Of course, having two rapsberry pi's in this circumstance is perhaps a bit redundant, but it would probably be a good idea anyway to isolate your device logic & control from our web-server to minimize any interference. We designed our C++ library to be as lightweight as possible, with as little chance as possible of interfering with the ventilator control circuit. 

You can view the [C++ library documentation on our wiki.](https://github.com/Open-Ventilator-Remote-Monitoring/docs/wiki/Serial-API:-Arduino)

The overall system architecture is here: 
[image]

ralphBellofattoSTX, 2020-04-12:
updated slides for sw arch...
[image]
[image]
[image]

ralphBellofattoSTX, 2020-04-15:
revised simulator, with rotary push button encoder.

[image]

ralphBellofattoSTX, 2020-04-15:
@alcohen for your review, one more revision, added leds to the simulator board and the rotoary encoder.
[image]

ralphBellofattoSTX, 2020-04-20:
closing this issue in favor of a Real Software architecture document:
https://github.com/JamVent/wiki/issues/31

closed 2020-04-20 by ralphBellofattoSTX