GlobalVent/wiki #25: QT Quick Proto by Joe Wild

ralphBellofattoSTX, 2020-04-16:
We would like a qt proto of the first screen of the jamvent system.

Done in either Qt Creator orVisualGDB.  Whichever is faster.

alcohen, 2020-04-16:
[image]

jwildhome, 2020-04-16:
Ok. Got it

ralphBellofattoSTX, 2020-04-17:
@jwildhome 

on the topic of QML and VisualGDB we got this answer from their support guys.  So.... i suspect this is going to be a qwidgets play...

I'll look into how to get us the complete 800x400 screen to work with.

> Hi,
> 
> Sorry, VisualGDB only supports the C/C++ debugging based on the GDB debugger. It does not support JavaScript/QML debugging. 
> 
> --
> Best regards,
>   Sysprogs Support 
>   View conversation thread online
>

jwildhome, 2020-04-17:
I agree that QWidgets is probably the direction to go and will try to focus my attention there.

As far as debugging QML/Javascript,, I think that has always been kind of weak, so I don't think we have been losing much there.  Once I've gotten QML integrated into a real application, I tend to keep the Javascript to a minimum and debug with log statements.

Maybe, this is because I tend to build and run things outside an IDE.  Maybe I will learn something new about GDB.

awalsh-personal, 2020-04-18:
There's a (crude) wireframe on gdrive https://drive.google.com/drive/folders/1yq_DNtv-JqbGgNawlf91kPG9mLDPDVk3?hl=en with agreed UI elements based on discussion with Bryan and Joost.  Feel free to email with any questions (for some reason slack notifications don't seem reliable yet for me).

jwildhome, 2020-04-20:
Here is an image of the Treatment screen that I mocked up in QWidgets.  I was able to build on WIndows (of course) and on my RPI3 as host.  I tried to keep this simple and just a few files to see how it looked and can be ported.

I struggled to get the VisualGDB x-compile environment set up.  Problems configuring/building Qt 5.12.8 to x-compile to install in the RPI.  Complained about incompatible versions of Open GL.  I understand that there is a Wiki page that I should read that may help with this.

Windows:
[image]

RPI:
[image]

ralphBellofattoSTX, 2020-04-25:
update from joe on slack:

>  8:21 AM
> Ralph,
> I added a new project /prototypes/JoesPrototypes/JamventDemoXCompile.
> This should build/run in our x-compile VGDB setup. At least it did for me.  Still using QCharts, so will need to install QCharts on RPI and sync sysroot/
> Run this on your RPI:
>   sudo apt install libqt5charts5 libqt5charts5-dev
> Saw your interface in jamctrllib/JamCtl.h.  Looks good. About how I was imagining it.
> I am planning to start with some prototyping with QCharts to make sure it roughly has the functionality we need.  And then I was going to clean up the the input widgets (bottom row) and output widgets (right column) to use with the JamCtrl.h interface. Does that sound reasonable?
> Probably won't work on this again until tomorrow morning.  As you may have noticed I try to get in a couple of hours of work in each morning before the house is awake.
> 
> Joe Wild  8:26 AM
> Forgot to mention.  I tried to clean up my project to not put build stuff in the source directory.
> Only partially successful.  I now put VisualGDB/ and CodeDB/ under ../Build.  So I can just delete the whole mess before committing to git.
> I could not figure out how to move VisualGDBCache/ so I updated our .gitignore. Hope that is OK.
> # visual Gdb
> #VisualGDB/
> VisualGDBCache/
> *.vgdbcmake.*.user
> CMakeLists.txt.old
> #CMakeFiles/
> #CMakeCache.txt (edited)

ralphBellofattoSTX, 2020-04-25:
@jwildhome 

And the first problem with picking up the build:
```
cmake ../../../JamventDemoXCompile -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DTOOLCHAIN_ROOT=C:/SysGCC/raspberry -DCMAKE_TOOLCHAIN_FILE=X:/github/prototypes/JoesPrototypes/JamventDemoXCompile/toolchain.cmake -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
-------------------------------------------------------------
Command exited with code -1
Executable: cmake
Arguments: ../../../JamventDemoXCompile -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DTOOLCHAIN_ROOT=C:/SysGCC/raspberry -DCMAKE_TOOLCHAIN_FILE=X:/github/prototypes/JoesPrototypes/JamventDemoXCompile/toolchain.cmake -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
Directory: X:\github\prototypes\JoesPrototypes\JamventDemoXCompile/../Build/VisualGDB/Debug
Command-line action failed
```

ralphBellofattoSTX, 2020-04-25:
this is a cmake project rather than a ninja so adding --verbose to the cmake options produces:
```
Running CMake: C:\Users\ralphbellofatto\AppData\Local\VisualGDB\CMake\bin\cmake.exe --verbose ../../../JamventDemoXCompile -G "Ninja" -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_COLOR_MAKEFILE=OFF -DTOOLCHAIN_ROOT=C:/SysGCC/raspberry -DCMAKE_TOOLCHAIN_FILE=X:/github/prototypes/JoesPrototypes/JamventDemoXCompile/toolchain.cmake -DCMAKE_MAKE_PROGRAM="C:/Program Files (x86)/Sysprogs/VisualGDB/ninja.exe"
CMake Error at C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/lib/arm-linux-gnueabihf/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5Charts" with
  any of the following names:

    Qt5ChartsConfig.cmake
    qt5charts-config.cmake

  Add the installation prefix of "Qt5Charts" to CMAKE_PREFIX_PATH or set
  "Qt5Charts_DIR" to a directory containing one of the above files.  If
  "Qt5Charts" provides a separate development package or SDK, be sure it has
  been installed.
```

ralphBellofattoSTX, 2020-04-25:
doing this:
```
sudo apt install libqt5charts5 libqt5charts5-dev
```

making sure synchronize sysroot is enabled.
[image]

So... where is the "build settings" page...

this is what is there:
[image]

ralphBellofattoSTX, 2020-04-25:
still get this anoying hand correct item.
```
#include "../../../../../JamventDemoXCompile/mainwidget.h"
```

Correct it, and build it...

add the DISPLAY=:0 environment option.

ralphBellofattoSTX, 2020-04-25:
a screen capture:
[image]

and a photo on the target device:

NOTE: the texturing is seems to be an artifact of the camera and does not show up on the screen itself
[image]

ralphBellofattoSTX, 2020-04-25:
@jwildhome ok, joe, now we need to get something that will show us fake telemetry.

Can you have something that would just graph a square wave that was updated lets say every 0.1 seconds with an additional data point and did the type of scrolling we had talked about in our monday meetigns?

jwildhome, 2020-04-26:
@ralphBellofattoSTX 
Updated prototypes/JoesPrototypes/JamventDemoXCompile.
Added a ChartWidget to hold all/most of the QChart stuff.  Added 3 instances to the main screen/widget to show a scatter, line, and spline graph.  Generate random values with a timer inside the Charts for now.

Once I removed scrolling and disabled QChart animations, I was able to update at 100ms. Will probably want to do some buffering or something.  Too fast for humans.

I had for force some font pixel sizes  for the QChart text as the RPI defaulted to larger fonts which did not fit as well.

Might want to consider putting fonts in the app to be safe. Best place would be the resource file, so we may have to fight that battle at some point.

keithel, 2020-04-28:
Keith Kyzivat of The Qt Company. I work Professional Services for them.
Since you are working with a Raspberry Pi, I think I'd recommend using QML and the scene graph - as you can take advantage of the GPU rendering OpenGLES2. The graphics performance should be enough for your needs then. If you use Widgets, you will only be software rendering, and the performance will probably not be enough.

The bad news is that there's really not great options for realtime plotting built-in to Qt right now.  If you were using Qt Widgets, I'd recommend using QWT, however, as I mentioned above, I don't think the performance will be good enough.  Qt Charts is not very good, and QWT is a good replacement.

With QML - plotting typically is a roll-your-own situation, where it's custom to the particular application.  I haven't done any of that myself, but I've seen some examples.  There is one project out there that aims to provide some reusable components that offer plotting though - so I'll list that, and it may be the right tool for the job:
https://www.arnorehn.de/blog/2014/12/26/quickplot-a-collection-of-native-qtquick-plot-items/

Regarding debugging into QML - you can use Qt Creator for that. Even if Qt Creator isn't used for the main build, you can use it after the fact for debugging into QML.  Write most of the business logic in C++, and just use QML for the display.

Another thing using QML - that leaves open the possibility of scaling this down to run on an MCU. Qt has a (commercial-only) MCU offering that we just released this past year. It is commercial, but given the situation with COVID-19, they may be willing to do something about that.  It is probably a good idea to start out with the RPi, but engineer things such that it's possible to scale down to use an MCU.

ralphBellofattoSTX, 2020-04-28:
my 2 cents...

This is looking more and more like we should be basing the graphing on QPainter, and just draw curved lines to connect the data points,,

once we get that, surround it by painting the legend around the edges.  I think that there will be much more time chewed up by trying to get qt to do this than to take a qpainter example and just make it draw what we want.

jwildhome, 2020-04-29:
Keith,  Thank you for commenting. Good to get your opinion. I agree about QML. We've started with QWidgets as that seemed the most practical to get our very basic screens prototyped.  Not quite ready to bring the QML framework yet.

Ralph, I will prototype a very simple widget that uses QPainter. I'm only going to draw lines at this point.  If I remember correctly, getting the curve for 2 points requires knowing the previous and next points (4 points all together), and a fair amount of math.

Then we can decide if we want to write write our own graphing widget, or can find a library that meets our needs and licensing requirements.  I think we are in for a boatload of work if we write our own (and maybe either way).

keithel, 2020-04-29:
Re graphing widget - take a look at QWT for your prototyping needs. It's pretty easy to work with, and the licensing should be compatible - it's licensed under a modified LGPL 2.1 license: https://qwt.sourceforge.io/qwtlicense.html

It may be performant enough for the prototype.

jwildhome, 2020-04-29:
Keith,  Thank you for the recommendation.  I will check it out.

jwildhome, 2020-04-29:
Updated JoesPrototypes/JamventDemoXCompile to show 1 QChart widget on the lower left
and 1 QPainter widget on lower right.  QPainter widget just draws simple lines and a blank (black) cursor.  Swap colors (red/green) each cycle to make update obvious.  Here is a picture:
[image]

ralphBellofattoSTX, 2020-04-29:
with sufficient data points, lets say 50 points every second, and the kind of curves we are taking about here, straight lines are probably ok.

alcohen, 2020-04-30:
Great Joe! Since I don’t have the time to build it, what’s your thought on the performance?

Also, in cases where there’s more than 1 or 2 samples per pixel column (this might be one case?) I’ve been able to speed things up by finding the max and min value for all the samples that map to a given pixel column, then drawing a vertical line between max and min.

jwildhome, 2020-04-30:
On performance, both the QChart and QPainter graphs seem to perform fine on a 100ms cycle.  My current implementation is sloppy with an independent timer controlling each graph.  I intend to update to drive all graphs with 1 timer.

The number of points that I am showing is very gross, 6 or 24 points per graph.  Just picked something that matched the QChart parameters.  Will try something more realistic.

I suspect I will have to massage the graph data before showing.  Thanks for the suggestion on how to do that.

jwildhome, 2020-05-01:
Updated JoesPrototypes/JamventDemoXCompile. Mostly cleanup.  Updating
all the graphs on 1 timer.  More product like sine wave.

Still updating at 100ms which seems visually fine.  Also fine at 50ms.
[image]

ralphBellofattoSTX, 2020-05-01:
the qpaint version seems more like what we want for display purposes,  so now we need a way to get legends on it...

awalsh-personal, 2020-05-01:
Looks good Joe!

awalsh-personal, 2020-05-01:
[sorry, wrong button]

jwildhome, 2020-05-05:
Updated JoesPrototypes/JamventDemoXCompile to show the QPainter
graph widget with Y Axis(vertical) and Y Axis Max labels.

I still have to figure out how to show the correct background for
the waveform. Wave painting method and background drawing conflict.
[image]

jwildhome, 2020-05-06:
Updated JoesPrototypes/JamventDemoXCompile to show the a white background for the wave graph, and updated the input widgets to be a label, input widget pair.  Layout and sizes still a little awkward on RPI. Will continue to polish.  Need to do the same (label and data pair) for the data display widgets in the right column.
[image]

alcohen, 2020-05-06:
Hey Joe, this is looking good. Can you clarify what you mean by " I still
have to figure out how to show the correct background for the waveform"?

alcohen, 2020-05-06:
> Waveform background problem was 2 back with black background. That should
> be solved now as I updated the background to be white, which I think is
> close enough for now.

Got it

jwildhome, 2020-05-18:
I added a new subdirectory/project under JoesPrototypes/VentGui/gui/

It has essentially the same functionality.  I restructured in the following
manner.
  gui/
    Main Controller, main.cpp

    common/widgets/
      common widgets

    common/resources/
      resource files, like images and fonts. (not used yet)

    pages/
      widgets and controllers for each page. Only Treatment page for now.

closed 2020-08-03 by jwildhome