GlobalVent/wiki #27: Labview License

ralphBellofattoSTX, 2020-04-17:
@JMathiszig-Lee 

We have been able to get some of the algorithms out of the labview implementation using a 45 day license that expires on May 19.2020. 

My understanding is that the Labview Algorithm and implementation is still being modified since hte last time i extracted at least some of the implementation from it, so it is likely we will need 1 licenced copy of it to continue the software work of translating the control algorithms to the RPI after that date.

jmriphagen, 2020-04-17:
@ralphBellofattoSTX .
Reached out and NI are on board and the licences will be no problem to renew for the length of the development and they possibly will get us a 6000 dac as well.

ralphBellofattoSTX, 2020-04-17:
@jmriphagen great, how do i get a key so i can apply to the installation?

jmriphagen, 2020-04-17:
Will be contacting them Monday for what we need (they emailed back within 2 hours after speaking to them on the phone). They might be able to supply people that can do stuff as well (if needed)

achortos, 2020-04-18:
If you need some 6000 DAQs, we have ~5 of them in the lab. We can use them for some of the first prototypes.

dcstraney, 2020-08-11:
Have since moved on from LabView to software running on the Pi

closed 2020-08-11 by dcstraney