GlobalVent/wiki #29: visualGDB prototype project structure.

ralphBellofattoSTX, 2020-04-18:
I would like to get a prototype project structure for jamvent under a single visualGDB project with the following structure.

* jamventProj -- top level jamvent project (mostly a container for subprojects).
  * jamvent -- jamvent ui and program main.
  * jamventhw -- jamvent hardware layer.
  * jamventsimlib -- jamvent simulator library code (portable to the hw simulator)
  * jamventsw_sim -- jamvent software simulator (replaces the jamvent hardware).

The design should be simple to re-arrange each of the subsystems, but it should compile and deploy at one touch...

ralphBellofattoSTX, 2020-04-18:
trial 1.
Create an application project that we will re-arrange.
[image]

[image]

[image]

ralphBellofattoSTX, 2020-04-18:
and then start copying files from the other qt Prototype we did before with the wizard.
```
now construct the new project...
```
cd jamventProtoProj01
mkdir jamvent
cp ../qtStarterAppCross/CMakeLists.txt jamvent
cp ../qtStarterAppCross/MainWindow.* jamvent
cp ../qtStarterAppCross/qtStarterAppCross.cpp  jamvent
```

and now our simulator framework (still in development)
```

comment out the executable.  we will add sub-projects as cmake files.
```
#add_executable(jamventProtoProj01 jamventProtoProj01.cpp)
#target_link_libraries(jamventProtoProj01 "${LIBRARIES_FROM_REFERENCES}")
```

```
cd jamventProtoProj01

```
[jamventProtoProj01]cp ../../cppJamventSim/src/CMakeLists.txt jamventsimlib/
[jamventProtoProj01]cp ../../cppJamventSim/src/*.cpp jamventsimlib/
[jamventProtoProj01]cp ../../cppJamventSim/src/*.h jamventsimlib/
```

create a util lib
```
makedir util

```

ralphBellofattoSTX, 2020-04-18:
copy the project:
```
[qtStarterApps]cp -r jamventProtoProj02 jamventProtoProj03
```
change the cmake and the .sin file.
.sin file change.
```
Project("{803FD0C6-D64E-4E16-9DC3-1DAEC859A3D2}") = "jamventProtoProj03", "jamventProtoProj03.vgdbcmake", "{98069C50-A92F-461A-BAD3-6A90BAA200FF}"
```

and the top level CMakeLists.txt file
```
project(jamventProtoProj03)
```

open and verify that we are ok.

ralphBellofattoSTX, 2020-04-18:
verify it builds, deploys and tests...

NOTE: it is not automatically copying the executable unless you tell it to debug the exe...

ralphBellofattoSTX, 2020-04-18:
lets see what we can get rid of.

move the VisualGDB directories out of the way.

project works ok if we don't have that...

```
[jamventProtoProj03]mkdir ../p3save
[jamventProtoProj03]mv VisualGDB* ../p3save/
```

ralphBellofattoSTX, 2020-04-18:
now trying CodeDB..
```
[jamventProtoProj03]mv CodeDB ../p3save/
```

And it also works...
Last thing to try:
```
[jamventProtoProj03]mv .vs ../p3save/
```

and that works too.

So the a prototype .gitignore is:
````
.vs/
CodeDB/
VisualGDB/
VisualGDBCache/

```

ralphBellofattoSTX, 2020-04-18:
test, 
commit them to git.
push the changes to repo.
remove the JamventProtoProj03 directory, 
and recheck it out...

```
git status
git add .
git status
git commit 
git push
cd ..
rm -rf jamventProtoProj03/
git checkout jamventProtoProj03
```

ralphBellofattoSTX, 2020-04-18:
cool, this is seems to be working, now
lets try to write some code...

more simulator stuff for now.

Then subsystem apis...

ralphBellofattoSTX, 2020-04-18:
tried to move us up one directory level and got surprised:

```

C:\SysGCC\raspberry\bin\arm-linux-gnueabihf-g++.exe --sysroot=C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_WIDGETS_LIB -Ijamvent -I../../jamvent -Ijamvent/jamvent_autogen/include -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5 -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtCore -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -g   -fPIC -MD -MT jamvent/CMakeFiles/jamvent.dir/jamvent_autogen/mocs_compilation.cpp.o -MF jamvent\CMakeFiles\jamvent.dir\jamvent_autogen\mocs_compilation.cpp.o.d -o jamvent/CMakeFiles/jamvent.dir/jamvent_autogen/mocs_compilation.cpp.o -c jamvent/jamvent_autogen/mocs_compilation.cpp
In file included from jamvent/jamvent_autogen/mocs_compilation.cpp:2:
jamvent/jamvent_autogen/EWIEGA46WW/moc_MainWindow.cpp:9:10: fatal error: ../../../../../jamvent/MainWindow.h: No such file or directory
 #include "../../../../../jamvent/MainWindow.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[4/10] Building CXX object jamventsimlib/testing/CMakeFiles/jsimTestConf.dir/testConf.cpp.o
```

So where did this come from

ralphBellofattoSTX, 2020-04-18:
so what is generating this:
```
./VisualGDB/Debug/jamvent/jamvent_autogen/EWIEGA46WW/moc_MainWindow.cpp:#include "../../../../../jamvent/MainWindow.h"
```

check out this:
```
./VisualGDB/Debug/jamvent/jamvent_autogen/EWIEGA46WW/moc_MainWindow.cpp:#include "../../../../../jamvent/MainWindow.h"
```

ralphBellofattoSTX, 2020-04-18:
this came from:
```
/****************************************************************************
** Meta object code from reading C++ file 'MainWindow.h'
**
** Created by: The Qt Meta Object Compiler version 67 (Qt 5.11.2)
**
** WARNING! All changes made in this file will be lost!
*****************************************************************************/
/****************************************************************************
** Meta object code from reading C++ file 'MainWindow.h'
**
** Created by: The Qt Meta Object Compiler version 67 (Qt 5.11.2)
**
** WARNING! All changes made in this file will be lost!
*****************************************************************************/

#include "../../../../../jamvent/MainWindow.h"
```

so moving the directory around with a qt project is, well interesting...

I think i had the same problem getting stephanies file to be able to use...

ralphBellofattoSTX, 2020-04-18:
moc_Mainwindow.cpp is generated, somewhere...

troubleshooting try the ninja.exe -v option:
[image]

ralphBellofattoSTX, 2020-04-18:
```
C:\PROGRA~2\Sysprogs\VISUAL~1/ninja.exe -v
[1/10] C:\SysGCC\raspberry\bin\arm-linux-gnueabihf-g++.exe --sysroot=C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot   -I../../utils -I../../jamventsim -g -MD -MT jamventsimlib/CMakeFiles/jamventsim.dir/GassimConst.cpp.o -MF jamventsimlib\CMakeFiles\jamventsim.dir\GassimConst.cpp.o.d -o jamventsimlib/CMakeFiles/jamventsim.dir/GassimConst.cpp.o -c ../../jamventsimlib/GassimConst.cpp
[2/10] C:\SysGCC\raspberry\bin\arm-linux-gnueabihf-g++.exe --sysroot=C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot   -I../../utils -I../../jamventsim -g -MD -MT jamventsimlib/CMakeFiles/jamventsim.dir/GassimConv.cpp.o -MF jamventsimlib\CMakeFiles\jamventsim.dir\GassimConv.cpp.o.d -o jamventsimlib/CMakeFiles/jamventsim.dir/GassimConv.cpp.o -c ../../jamventsimlib/GassimConv.cpp
[3/10] cmd.exe /C "cd . && C:\Users\ralphbellofatto\AppData\Local\VisualGDB\CMake\bin\cmake.exe -E remove jamventsimlib\libjamventsim.a && C:\SysGCC\raspberry\bin\arm-linux-gnueabihf-ar.exe qc jamventsimlib\libjamventsim.a  jamventsimlib/CMakeFiles/jamventsim.dir/GassimConst.cpp.o jamventsimlib/CMakeFiles/jamventsim.dir/GassimConv.cpp.o && C:\SysGCC\raspberry\bin\arm-linux-gnueabihf-ranlib.exe jamventsimlib\libjamventsim.a && cd ."
[4/10] cmd.exe /C "cd /D X:\github\prototypes\jamventProto\VisualGDB\Debug\jamvent && C:\Users\ralphbellofatto\AppData\Local\VisualGDB\CMake\bin\cmake.exe -E cmake_autogen X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent/CMakeFiles/jamvent_autogen.dir/AutogenInfo.cmake DEBUG"
[5/10] C:\SysGCC\raspberry\bin\arm-linux-gnueabihf-g++.exe --sysroot=C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_WIDGETS_LIB -Ijamvent -I../../jamvent -Ijamvent/jamvent_autogen/include -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5 -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtCore -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -g   -fPIC -MD -MT jamvent/CMakeFiles/jamvent.dir/jamvent_autogen/mocs_compilation.cpp.o -MF jamvent\CMakeFiles\jamvent.dir\jamvent_autogen\mocs_compilation.cpp.o.d -o jamvent/CMakeFiles/jamvent.dir/jamvent_autogen/mocs_compilation.cpp.o -c jamvent/jamvent_autogen/mocs_compilation.cpp
FAILED: jamvent/CMakeFiles/jamvent.dir/jamvent_autogen/mocs_compilation.cpp.o 
C:\SysGCC\raspberry\bin\arm-linux-gnueabihf-g++.exe --sysroot=C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_WIDGETS_LIB -Ijamvent -I../../jamvent -Ijamvent/jamvent_autogen/include -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5 -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtCore -isystem C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -g   -fPIC -MD -MT jamvent/CMakeFiles/jamvent.dir/jamvent_autogen/mocs_compilation.cpp.o -MF jamvent\CMakeFiles\jamvent.dir\jamvent_autogen\mocs_compilation.cpp.o.d -o jamvent/CMakeFiles/jamvent.dir/jamvent_autogen/mocs_compilation.cpp.o -c jamvent/jamvent_autogen/mocs_compilation.cpp
In file included from jamvent/jamvent_autogen/mocs_compilation.cpp:2:
jamvent/jamvent_autogen/EWIEGA46WW/moc_MainWindow.cpp:9:10: fatal error: ../../../../../jamvent/MainWindow.h: No such file or directory
 #include "../../../../../jamvent/MainWindow.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

no joy here..

when does the moc file get created...

close the project:
```
[jamventProto]rm -rf CodeDB/ VisualGDB* .vs
```

then start the project up and see what gets generated...

ralphBellofattoSTX, 2020-04-18:
open the project and let it read in the cmake files.
no moc generated yet..
```
[jamventProto]find . -name 'moc*'
```

copy the files to jamventProto.ref
```
[prototypes]cp -r  jamventProto jamventProto.ref
```

open the project.
copy to ref2
```
[prototypes]cp -r  jamventProto jamventProto.ref2
```

then do a build and see when the moc shows up.

ralphBellofattoSTX, 2020-04-18:
list of the new files when building:
```
Folder Compare
Produced: 4/18/20 5:40:10 PM

Mode:  Differences
Ignoring Unimportant
Left base folder: /Volumes/ct-ng/covid-19/github/prototypes/jamventProto.ref2
Right base folder: /Volumes/ct-ng/covid-19/github/prototypes/jamventProto

Left Orphan Files (0)                                                            Size    Modified
------------------------------------------------------------------------------------------------------------

Right Orphan Files (13)                                                          Size    Modified
------------------------------------------------------------------------------------------------------------
VisualGDB/Debug/jamvent/CMakeFiles/jamvent.dir/qtStarterAppCross.cpp.o           639,296 4/18/20, 5:38:05 PM
VisualGDB/Debug/jamvent/CMakeFiles/jamvent_autogen.dir/AutogenOldSettings.txt        138 4/18/20, 5:38:02 PM
VisualGDB/Debug/jamvent/CMakeFiles/jamvent_autogen.dir/ParseCache.txt                268 4/18/20, 5:38:02 PM
VisualGDB/Debug/jamvent/jamvent_autogen/EWIEGA46WW/moc_MainWindow.cpp              3,516 4/18/20, 5:38:02 PM
VisualGDB/Debug/jamvent/jamvent_autogen/include/ui_MainWindow.h                    2,606 4/18/20, 5:38:02 PM
VisualGDB/Debug/jamvent/jamvent_autogen/moc_predefs.h                             16,118 4/18/20, 5:38:01 PM
VisualGDB/Debug/jamvent/jamvent_autogen/mocs_compilation.cpp                         101 4/18/20, 5:38:02 PM
VisualGDB/Debug/jamventsimlib/CMakeFiles/jamventsim.dir/GassimConst.cpp.o          1,076 4/18/20, 5:38:01 PM
VisualGDB/Debug/jamventsimlib/CMakeFiles/jamventsim.dir/GassimConv.cpp.o           1,072 4/18/20, 5:38:01 PM
VisualGDB/Debug/jamventsimlib/testing/CMakeFiles/jsimTestConf.dir/testConf.cpp.o 213,028 4/18/20, 5:38:03 PM
VisualGDB/Debug/jamventsimlib/libjamventsim.a                                      2,438 4/18/20, 5:38:01 PM
VisualGDB/Debug/.ninja_deps                                                       35,388 4/18/20, 5:38:05 PM
VisualGDB/Debug/.ninja_log                                                           585 4/18/20, 5:38:05 PM
------------------------------------------------------------------------------------------------------------

Left Newer Files (0)                                                             Size    Modified
------------------------------------------------------------------------------------------------------------

Right Newer Files (0)                                                            Size    Modified
------------------------------------------------------------------------------------------------------------

Difference Files (1)                                                             Size    Modified
------------------------------------------------------------------------------------------------------------
.vs/jamventProto/v16/.suo                                                         14,848 4/18/20, 5:35:12 PM
                                                                                  18,944 4/18/20, 5:38:01 PM
------------------------------------------------------------------------------------------------------------
```

ralphBellofattoSTX, 2020-04-18:
so who generates the moc_Mainwindow.cpp file and why is it being generated like this?

this shows up here:
```
[prototypes]find  jamventProto.ref2/ -type f | xargs grep moc_
jamventProto.ref2//VisualGDB/Debug/jamvent/CMakeFiles/jamvent_autogen.dir/AutogenInfo.cmake:set(AM_HEADERS_BUILD_PATHS "EWIEGA46WW/moc_MainWindow.cpp")
```

so, who is doing this and where and why?

ralphBellofattoSTX, 2020-04-18:
hand correcting that line to this:
```
#include "../../../jamvent/MainWindow.h"
```

got it to build: so remove ../../

how is it figuring out where that should have been.

ralphBellofattoSTX, 2020-04-18:
now here is a funny.

I hand edited the file, deleteed the entire three directories
CodeDB Visual*

restarted the studio 
and then rebuild it and it worked

so...hm... what was different...

ralphBellofattoSTX, 2020-04-18:
this is the step that generates the wrong moc_Mainwindow.cpp incluide.

```
[4/10] cmd.exe /C "cd /D X:\github\prototypes\jamventProto\VisualGDB\Debug\jamvent && C:\Users\ralphbellofatto\AppData\Local\VisualGDB\CMake\bin\cmake.exe -E cmake_autogen X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent/CMakeFiles/jamvent_autogen.dir/AutogenInfo.cmake DEBUG"
```

and here is the contents of the AutogenInfo.cmake
```
# Meta
set(AM_MULTI_CONFIG "FALSE")
set(AM_PARALLEL "2")
set(AM_VERBOSITY "")
# Directories
set(AM_CMAKE_SOURCE_DIR "X:/github/prototypes/jamventProto")
set(AM_CMAKE_BINARY_DIR "X:/github/prototypes/jamventProto/VisualGDB/Debug")
set(AM_CMAKE_CURRENT_SOURCE_DIR "X:/github/prototypes/jamventProto/jamvent")
set(AM_CMAKE_CURRENT_BINARY_DIR "X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent")
set(AM_CMAKE_INCLUDE_DIRECTORIES_PROJECT_BEFORE "")
set(AM_BUILD_DIR "X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent/jamvent_autogen")
set(AM_INCLUDE_DIR "X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent/jamvent_autogen/include")
# Qt
set(AM_QT_VERSION_MAJOR 5)
set(AM_QT_MOC_EXECUTABLE "C:/SysGCC/raspberry/Qt/v5-CMake/moc.exe")
set(AM_QT_UIC_EXECUTABLE "C:/SysGCC/raspberry/Qt/v5-CMake/uic.exe")
# Files
set(AM_CMAKE_EXECUTABLE "C:/Users/ralphbellofatto/AppData/Local/VisualGDB/CMake/bin/cmake.exe")
set(AM_SETTINGS_FILE "X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent/CMakeFiles/jamvent_autogen.dir/AutogenOldSettings.txt")
set(AM_PARSE_CACHE_FILE "X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent/CMakeFiles/jamvent_autogen.dir/ParseCache.txt")
set(AM_HEADERS "X:/github/prototypes/jamventProto/jamvent/MainWindow.h")
set(AM_HEADERS_FLAGS "MU")
set(AM_HEADERS_BUILD_PATHS "EWIEGA46WW/moc_MainWindow.cpp")
set(AM_SOURCES "X:/github/prototypes/jamventProto/jamvent/MainWindow.cpp;X:/github/prototypes/jamventProto/jamvent/qtStarterAppCross.cpp")
set(AM_SOURCES_FLAGS "MU;MU")
# MOC settings
set(AM_MOC_SKIP "")
set(AM_MOC_DEFINITIONS "QT_CORE_LIB;QT_GUI_LIB;QT_WIDGETS_LIB")
set(AM_MOC_INCLUDES "X:/github/prototypes/jamventProto/VisualGDB/Debug/jamvent;X:/github/prototypes/jamventProto/jamvent;C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5;C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtWidgets;C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtGui;C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf/qt5/QtCore;C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++;C:/SysGCC/raspberry/arm-linux-gnueabihf/include/c++/8;C:/SysGCC/raspberry/arm-linux-gnueabihf/include/c++/8/backward;C:/SysGCC/raspberry/lib/gcc/arm-linux-gnueabihf/8/include;C:/SysGCC/raspberry/lib/gcc/arm-linux-gnueabihf/8/include-fixed;C:/SysGCC/raspberry/arm-linux-gnueabihf/include;C:/SysGCC/raspberry/arm-linux-gnueabihf/include/arm-linux-gnueabihf/c++/8;C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include/arm-linux-gnueabihf;C:/SysGCC/raspberry/arm-linux-gnueabihf/sysroot/usr/include")
set(AM_MOC_OPTIONS "")
set(AM_MOC_RELAXED_MODE "")
set(AM_MOC_MACRO_NAMES "Q_OBJECT;Q_GADGET;Q_NAMESPACE")
set(AM_MOC_DEPEND_FILTERS "")
set(AM_MOC_PREDEFS_CMD "C:/SysGCC/raspberry/bin/arm-linux-gnueabihf-g++.exe;-dM;-E;-c;C:/Users/ralphbellofatto/AppData/Local/VisualGDB/CMake/share/cmake-3.15/Modules/CMakeCXXCompilerABI.cpp")
# UIC settings
set(AM_UIC_SKIP "")
set(AM_UIC_TARGET_OPTIONS "")
set(AM_UIC_OPTIONS_FILES "")
set(AM_UIC_OPTIONS_OPTIONS "")
set(AM_UIC_SEARCH_PATHS "")
```

ralphBellofattoSTX, 2020-04-18:
Trying an experiment, just move the jamventProtoProj03 up one level, do a clean and rebuld and see what happens.

ralphBellofattoSTX, 2020-04-19:
and here is something curious about cmake -E...
does not list cmake_autogen as a command... hm...

```
Usage: C:\Users\ralphbellofatto\AppData\Local\VisualGDB\CMake\bin\cmake.exe -E <command> [arguments...]
Available commands:
  capabilities              - Report capabilities built into cmake in JSON format
  chdir dir cmd [args...]   - run command in a given directory
  compare_files [--ignore-eol] file1 file2
                              - check if file1 is same as file2
  copy <file>... destination  - copy files to destination (either file or directory)
  copy_directory <dir>... destination   - copy content of <dir>... directories to 'destination' directory
  copy_if_different <file>... destination  - copy files if it has changed
  echo [<string>...]        - displays arguments as text
  echo_append [<string>...] - displays arguments as text but no new line
  env [--unset=NAME]... [NAME=VALUE]... COMMAND [ARG]...
                            - run command in a modified environment
  environment               - display the current environment
  make_directory <dir>...   - create parent and <dir> directories
  md5sum <file>...          - create MD5 checksum of files
  sha1sum <file>...         - create SHA1 checksum of files
  sha224sum <file>...       - create SHA224 checksum of files
  sha256sum <file>...       - create SHA256 checksum of files
  sha384sum <file>...       - create SHA384 checksum of files
  sha512sum <file>...       - create SHA512 checksum of files
  remove [-f] <file>...     - remove the file(s), use -f to force it
  remove_directory <dir>... - remove directories and their contents
  rename oldname newname    - rename a file or directory (on one volume)
  server                    - start cmake in server mode
  sleep <number>...         - sleep for given number of seconds
  tar [cxt][vf][zjJ] file.tar [file/dir1 file/dir2 ...]
                            - create or extract a tar or zip archive
  time command [args...]    - run command and display elapsed time
  touch <file>...           - touch a <file>.
  touch_nocreate <file>...  - touch a <file> but do not create it.
  create_symlink old new    - create a symbolic link new -> old
Available on Windows only:
  delete_regv key           - delete registry value
  env_vs8_wince sdkname     - displays a batch file which sets the environment for the provided Windows CE SDK installed in VS2005
  env_vs9_wince sdkname     - displays a batch file which sets the environment for the provided Windows CE SDK installed in VS2008
  write_regv key value      - write registry value
```

ralphBellofattoSTX, 2020-04-19:
simple experiment.

create a directory hierachy:
```
[scratch]tree t1
t1
└── t2
    └── t3
```

```
create a qt app in this hierachy:
t2/t2/t3/LinuxProject1
```

And this worked...

hm....

* build
* test.
* clean

then do mv t1/t2/t3/qtTest t1
and rebuild and see what happens...

ralphBellofattoSTX, 2020-04-19:
now lets remove the files i think are non critical..

and now we get the problem
```
In file included from LinuxProject1_autogen/mocs_compilation.cpp:2:
LinuxProject1_autogen/EWIEGA46WW/moc_MainWindow.cpp:9:10: fatal error: ../../../../MainWindow.h: No such file or directory
 #include "../../../../MainWindow.h"
```

ralphBellofattoSTX, 2020-04-19:
repeating the experiment this time creating the same project in the same place.

and now i get exactly the same answer without moving it:
```

In file included from LinuxProject1_autogen/mocs_compilation.cpp:2:
LinuxProject1_autogen/EWIEGA46WW/moc_MainWindow.cpp:9:10: fatal error: ../../../../MainWindow.h: No such file or directory
 #include "../../../../MainWindow.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

and the same manual correction:
```
 #include "../../../../MainWindow.h"
```

Change to 
```
 #include "../../MainWindow.h"
```
and it worked...

jwildhome, 2020-08-11:
I think this is all in place and we can close this issue.

closed 2020-08-11 by carlyjb