GlobalVent/wiki #30: Write header files for subsystem apis.

ralphBellofattoSTX, 2020-04-18:
These header files will become part of a software design document.

It will be easier to prototype the headers by coding them as then they will be syntaicially correct.

Then these can go into a software design document with block level diagrams...

in broad strokes we are after this:
[image]

So we need a decsion of who is in charge, the ui thread or the control thread.

and we need the following api's defined at least going top down:

* JamventControlMgr -- jam vent control manager interface from the UI.
* JamventHwMgr -- jamvent hardware manager 

also -- break down the inside interfaces for the hardware manager.
* JamventHwImpl -- actual hardware implementation.
* JamventHwSimImpl -- simulated hardware impelmentation.

ralphBellofattoSTX, 2020-04-24:
@jwild 

i added an initial control manager header and checked it in.

Subject to further refinements, but please review.

ralphBellofattoSTX, 2020-05-25:
still need a header file for the harware layer...

carlyjb, 2020-05-27:
Ralph is this an open to do for you or a blocking task that you're waiting on someone for?

ralphBellofattoSTX, 2020-05-27:
we need the proof of concept on each of the io methods we are going to use.

* GPIO input.
* GPIO output.
* i2c input.

Before any meaning full class spec can be written.
I believe you created tasks for those today.