GlobalVent/wiki #31: software architecture.

ralphBellofattoSTX, 2020-04-20:
This issue tracks the development and revisions of the software architecture.

ralphBellofattoSTX, 2020-04-20:
software Design document version 3.
[sw_desgin_03.docx](https://github.com/JamVent/wiki/files/4504964/sw_desgin_03.docx)

and power point slides of same.
[sw_design_03.pptx](https://github.com/JamVent/wiki/files/4504966/sw_design_03.pptx)

awalsh-personal, 2020-06-11:
@ralphBellofattoSTX the SW architecture doc looks like a really good start.  some of the sections that you include would go into the SW development plan rather than the architecture doc, and are still essential for a regulatory filing.  A draft TOC for the design history file/technical file (all the quality records needed for a filing in US or EU) here https://www.dropbox.com/s/9kh8p1m91u3w4ik/DHF%20Table%20of%20Contents.xlsx?dl=0. I will post some doc templates for SW development plan, architecture, and detailed design documents sometime this week - they may be helpful as references.