GlobalVent/wiki #35: program the Barth STX-0600 device for the Jamvent prototype.

ralphBellofattoSTX, 2020-04-23:
we need the barth STX-0600 device to be programmed with the following:

```
in1 ---> out1
in2 ---> out2
in3---> out3
in4---> out4
```

Barth home page: https://barth-elektronik.com/

[stg-600-manual.pdf](https://github.com/JamVent/wiki/files/4519492/stg-600-manual.pdf)

MICON manual for programming this puppy.
[stg-micon-prog-manual.pdf](https://github.com/JamVent/wiki/files/4519493/stg-micon-prog-manual.pdf)

note the stx-0600 is an obsolete part, superceeded by other models.

ralphBellofattoSTX, 2020-04-23:
downloaded the software here:
https://micon-l.de/downloads.html?file=files/miCon-L/miCon-L%20Installationen/miCon-L_EN_V3.8.zip

this file contains an exe file, so it is windows only.

work under the vm...

ralphBellofattoSTX, 2020-04-23:
the usb device was not happy running on a vm. try using a real windows box...

ralphBellofattoSTX, 2020-04-23:
ok, i managed to do an initial program, however, i don't have the values of the hysteresis to plug in here:
I programmed 4 inputs mapping to 4 outputs.

[image]

In1 and in2 are digital inputs, so i assume i can just map them out.

However, inputs 3-9 are analog inputs, so i need to do some hysterisis to convert them to digital out.  However, the programming guide is a bit sparse...

So if there is someone who has the program file for this, so i can just load it, that would be a really BIG help...

ralphBellofattoSTX, 2020-04-24:
current program now is:
[image]

Verified with ohm meter.

alcohen, 2020-04-24:
Did the UK team supply their program?

closed 2020-04-24 by ralphBellofattoSTX

ralphBellofattoSTX, 2020-04-25:
this came off of a video:

[image]

https://www.youtube.com/watch?v=UvlAGWJBmJ8&t=1s