GlobalVent/wiki #38: Reading pressure sensors with I2C

carlyjb, 2020-05-10:


stephmusinsky, 2020-05-17:
Changing I2C sensor. Realized I had to order 3.3 Digilent PowerBrick (only got 5V one and no power supply at home :( )
Going to work on modifying code for new sensor today so can test when that arrives

stephmusinsky, 2020-05-24:
[image]

stephmusinsky, 2020-05-24:
[image]

stephmusinsky, 2020-05-24:
Looks to get ACK for reset sequence- going to go through the 5 stages for initialization

stephmusinsky, 2020-05-24:
[image]
currently- will pick up tomorrow

stephmusinsky, 2020-05-24:
don't do reset- having a little transition issue but working on it

stephmusinsky, 2020-05-24:
[image]

stephmusinsky, 2020-05-24:
got a nice sequence- reading the pressure offset from here
[image]

stephmusinsky, 2020-05-24:
Working on getting right wait time to poll ~10ms for pressure reading. Subtract from offset

stephmusinsky, 2020-05-24:
[image]
D6 high shows reading in offset value correctly

stephmusinsky, 2020-05-24:
get reading as well... but theres more math than expected to get to actual pressure value

stephmusinsky, 2020-05-25:
The math isn't actually so bad but we will probably need to go up in CPLD. Confirmed reading correctly (pressure- 1012.775 mbar)

closed 2020-05-25 by stephmusinsky