GlobalVent/wiki #39: UART comms with RasPi

carlyjb, 2020-05-10:


stephmusinsky, 2020-05-17:
Testing Transmit and looks right!
Wrote up quick arduino uart to send to cpld to test receive

stephmusinsky, 2020-05-17:
Modified some UART code online to include even parity

stephmusinsky, 2020-05-17:
w/ arduino and logic analyzer: Arduino reads on rx and repeats it back out on its tx. Looks good.
Going to do same test backwards

stephmusinsky, 2020-05-18:
RX sim works but some issues with arduino -> rx

stephmusinsky, 2020-05-18:
[image]

Looks to be working!! CPLD sends then Arduino receives, delays, sends back. Then CPLD receives and resends.

closed 2020-05-18 by stephmusinsky

ralphBellofattoSTX, 2020-05-18:
What was the test setup that showed we can communicate with the RPI?