GlobalVent/wiki #42: Build/Flash JamventSimLib on Photon

carlyjb, 2020-05-10:


ralphBellofattoSTX, 2020-05-12:
@ajo27

I updated the source code in prototypes/jamventProto.

ralphBellofattoSTX, 2020-05-16:
working out the details of creating a library, any library first for the photon using VSstudio.

installed the workbench for vsCode:
https://docs.particle.io/tutorials/developer-tools/workbench/#working-with-particle-projects

Following the directions here to create a toy library to go with the toy app and demonstrate that we can run generic c++ code...

ralphBellofattoSTX, 2020-05-16:
the readme has directions to compile it from the command line.

so it looks like library support needs to go through other gorpage...

as it does not seem to set up a vscode project stuff...

particle compile examples/usage

and it promptly complains about:
```
PS X:\github\prototypes\toyParticleLib> particle compile examples/usage
Compile failed: Target device examples/usage is not valid
        eg. particle compile core xxx
        eg. particle compile photon xxx
```

the ide has the same issue, and trying to select a device asks about a project. properties file which does not exist after creating the library.

ralphBellofattoSTX, 2020-05-16:
one can create a project properties, the first toy project only had a name in it so.
lets try:
```
[toyParticleLib]cat project.properties 
name=toyParticleLib
```
Then try to set device...

interestingly enough, the command went through but did not change anyting in the properties file.
those values went in the .vscode
```
[toyParticleLib]cat .vscode/settings.json 
{
    "particle.targetDevice": "particle01",
    "particle.firmwareVersion": "1.5.1",
    "particle.targetPlatform": "photon"
}[toyParticleLib]
```

ralphBellofattoSTX, 2020-05-16:
and we get this:
```
:::: COMPILING APPLICATION

cc1plus.exe: fatal error: x:/github/prototypes/toyParticleLib//src/toyParticleLib.cpp: No such file or directory
compilation terminated.
make[3]: *** [../build/module.mk:274: ../build/target/user/platform-6-m/toyParticleLib/src/toyParticleLib.o] Error 1
make[2]: *** [../../../build/recurse.mk:12: user] Error 2
make[1]: *** [../build/recurse.mk:12: modules/photon/user-part] Error 2
make: *** [C:\Users\ralphbellofatto\.particle\toolchains\buildscripts\1.9.2\Makefile:54: compile-user] Error 2
The terminal process terminated with exit code: 2

Press any key to close the terminal.
```

I'm wondering if there is some issue with the camelhump file names here...

lets try the library again, this time, with all lc names...

ralphBellofattoSTX, 2020-05-16:
after a hand edit of all the files generated and changing file names ot lower case,

Loading up the file indicated that it said the project was dirty, and asked to clean it
* clicked the clean button and it cleaned successfully:

```
*** CLEANED SUCCESSFULLY ***
```

and also came up with the particle workbench welcome as well.

clicked on the compile local and now am waiting for it to do its thing, it seems to take its time

make.exe appears to be doing something it is just quiet about it...

ralphBellofattoSTX, 2020-05-16:
took a while but it now says it is compiled successfully.
```
*** COMPILED SUCCESSFULLY ***

Press any key to close the terminal.
```

trying to flash the resulting example.ino file.

Off hand, it really looks like it has a headache with files with mixed case at least when on windows.

ralphBellofattoSTX, 2020-05-16:
supposedly there is serial printlin's happening on this.
```
 */
void ToyParticleLib::process()
{
    // do something useful
    Serial.println("called process");
    doit();
}

```
```
void setup() {
    // Call functions on initialized library objects that require hardware
    toyParticleLib.begin();
}

void loop() {
    // Use the library's initialized objects and functions
    toyParticleLib.process();
}
```
so this should blatther, but where does the output go?

ralphBellofattoSTX, 2020-05-16:
and we don't see anything out the serial monitor...
```
PS X:\github\prototypes\toyparticlelib> particle serial monitor
Opening serial monitor for com port: "COM5"
Serial monitor opened successfully:
```

serial monitor works when we embed some printlns in our original toy tests.

ralphBellofattoSTX, 2020-05-16:
added just one file to the library from the gassim area and it blew up with:
```
  /cygdrive/c/Users/ralphbellofatto/.particle/toolchains/deviceOS/1.5.1/modules/photon/system-part1/makefile /cygdrive/c/Users/ralphbellofatto/.particle/toolchains/deviceOS/1.5.1/modules/photon/system-part2/makefile /cygdrive/c/Users/ralphbellofatto/.particle/toolchains/deviceOS/1.5.1/modules/photon/user-part/makefile
   text    data     bss     dec     hex filename
 238292     212     424  238928   3a550 ../../../build/target/system-part1/platform-6-m/system-part1.elf
   text    data     bss     dec     hex filename
 259024    2420   40000  301444   49984 ../../../build/target/system-part2/platform-6-m/system-part2.elf
cc1plus.exe: fatal error: x:/github/prototypes/toyparticlelib//src/GassimConst.cpp: No such file or directory
compilation terminated.
make[3]: *** [../build/module.mk:274: ../build/target/user/platform-6-m/toyparticlelib/src/GassimConst.o] Error 1
make[2]: *** [../../../build/recurse.mk:12: user] Error 2
make[1]: *** [makefile:86: /cygdrive/c/Users/ralphbellofatto/.particle/toolchains/deviceOS/1.5.1/modules/photon/user-part/makefile] Error 2
make: *** [C:\Users\ralphbellofatto\.particle\toolchains\buildscripts\1.9.2\Makefile:60: compile-all] Error 2
The terminal process terminated with exit code: 2
```

there is not much there...

ralphBellofattoSTX, 2020-05-16:
changed the file case:
```
[src]mv GassimConst.h gassimconst.h
[src]mv GassimConst.cpp gassimconst.cpp
```

and it compiled successfully... hm...

ralphBellofattoSTX, 2020-05-16:
trying now to add gassim libraries to 
toyparticle01 in lib/gassimlib/src

move it into the directory struct and see if it picks it up...

or do we need to add more gorp.

ralphBellofattoSTX, 2020-05-16:
this time, it did find the files even without changing the file case.

Moving some other modules, next, GassimConv.*

ralphBellofattoSTX, 2020-05-16:
trying this around any header we may want to have for debugging elsewhere...
```
#ifndef defined(SPARK_CORE)
#include <iostream>
#endif
```

NOTE: we might want to feed a ostream object to the device for debug telemetry...

ralphBellofattoSTX, 2020-05-16:
successfully add the librarry to toyparticle01 in the prototypes area.

Next test would be actually calling it in a loop and haveing it do somethng we can detect...

Now, what do we have that would give us some visual indicator of what is going on.

* Serial debug.
* Leds of some sort (adding leds and 4k resistors to a simple breadboard).
* something connected to a pwm output that can react to the pulse width
  * a servo motor?  (i have a whole back of them), how many pwm outputs do we have?

ajo27, 2020-05-17:
SOLVED: Successfully compiled Ralph's library in Particle Workbench and via cloud compiler using Particle Dev

-After running particle library create to create a library for this code
-Take testModel00.cpp and rename to example.ino (needs a .ino extension), place in subfolder /examples/usage
-Built in VS Code/Workbench by selecting Particle: Create new project from command palette and then copying in library that I was already working with in Particle Dev
-Compiler error that "section `.text' will not fit in region `APP_FLASH' particle" -- overflowed by 20906 bytes -- SOLVED - all of the extra includes were eating up memory, namely <iostream>, commented out all references. After this compile was successful using local compiler in Particle Workbench and with the cloud compiler in Particle Dev.

-Flashed my edited version of his library to Photon and we get timestamps at each loop/model step iteration, and a "Test: 1" (Passed) after the 5 seconds of running the model is up. Need to commit this code to repo

ralphBellofattoSTX, 2020-05-17:
So, what do you think the next step is in getting this to work.  The testmodel00 does not do much more than make sure it compiles and pretty much sit there.

If you look testmodel01, that actually has a cycle of the data.  If you could do a continuous loop instead of 5 times through the valve sequence, i think that would demonstrate it actually working.  

Then you would need to try to figure out a way to get evidence that it is actually working in its new home.

carlyjb, 2020-05-18:
Code can be loaded & called.
Currently doesn't do anything, that would be next steps.

Next:
demo using serial output of waveforms to blindly toggle vales open & closed

closed 2020-05-18 by carlyjb