GlobalVent/wiki #52: Joe/Joost collab on display

carlyjb, 2020-05-11:


carlyjb, 2020-05-27:
done treatment screen
need to discuss other screens

jwildhome, 2020-06-05:
Joost and I met yesterday to discuss the UI screens and behavior.
Here are my notes.

As far as navigation behavior and knob events, agreed that starting
with the simple scheme that we get from Qt for free would be an
adequate start.  Simple behavior is: knob press to tab between
focusable elements, knob click right/left to increase/decrease input
element values and to select a button.

Full navigation/knob events could be fairly involved.  Will need to
move forward and back on the screen's focus list.  Will need to keep
track of state for focusable elements and show visually. Elements can
be in no focus, focus, focus and selected states.  Also, we will need
a scheme to confirm actions to reduce unintended actions.  I have a
rough idea of what we want to do here, but will need to spell out in
detail at some point.

There has been discussion of entering patient data of age, height,
weight, and sex.  These values could be used to set initial
parameters.  I don't think these are necessary for our early
prototypes.  Probably can just pick a reasonable set of initial
parameters.

As an aside, Al suggested we might want a settings/json that the GUI
would use for configuration.  So might want something like Config.json
that holds config parameters, like graph data (titles, axis labels,
ranges, ...).  And another file, Patient.json that holds this patient
data.

Joost suggested that it would be useful to compute and display a
target volume on the GUI.

Looked at the UI for another instrument as an example.  Here
are Joost's suggestions.
  * Prefers a simple style. gray background, green input widgets, light
    green display widgets.
  * Prefers units as part of label, "PEEP (cmH2O)" for input and display
    widgets.
  * Better to make input/display widgets as large as possible with
    as large a font as possible, so can read from a distance.

Start FiO2 at 100% as default. Is safe.

The Valve A-D buttons are not needed on the real UI. Only have now for
demonstration purposes.

Need at start up/power up screen.  Need to show that start up
diagnostics succeed or fail. For now, could just display a start up
text log and a start up status of success/fail.  Eventually, may want
sn option to skip the self test as these devices are tested regularly.

Reasonable startup paramters are 500-600 ml for volume and 10-12
breaths/minute.

Here is roughly what the input widget looked like on the device we
were discussing.  Not sure about "kPa" is right unit as hard to view
image.

Pmax    Vr    freq   T/INSP  PEEP
kPa     ml    1/min  sec     kPa
40      600   12     1.7     5

Typical values are 12 breaths/minute, volume of 600 ml.

There is also a pressure support mode, where the ventilator acts like
power steering to assist an already breathing patient.  This is not
planned to early releases.

Joost suggested that a tabbed interface may be a good or accepted approach
to switch between views/screens.

So at this point, I think I understand enough to extend the GUI for
our early integrations.

Here are what I think are some if the 1st tasks for the GUI.
  1. Add a skeleton Power Up/Start Up screen.
  2. Add a settings config file (Config.json) so can change basic
     GUI parameters without rebuilding.
  3. May also want to add a patient data config file, Patient.json

jwildhome, 2020-06-05:
I consider this done

jwildhome, 2020-06-05:
Oops. Closed by accident. Re-open and move to done.

ralphBellofattoSTX, 2020-06-05:
If the configuration information is all written by the program and never hand edited, json is a good choice.

If it is designed for human edit and input for program set, then I would recommend using Yaml.
https://yaml.org/

it can represent all the things json can represent and is easier to maintain.  Also important for human editing, you can have comments.  You can't have that with json.

closed 2020-06-05 by carlyjb

jwildhome, 2020-06-06:
I have not used YAML.  It seems like a nicer syntax for us humans.  My only concern is that it would require another library, where there is JSON support in Qt Core.

I am thinking of this more as a short-term solution.  In a product, it gives us another vulnerability and increases what we have to test/verify.

ralphBellofattoSTX, 2020-06-06:
there is a libyaml in the raspbain distribution.

```
pi@rpi02:~ $ apt-cache policy libyaml-cpp-dev
libyaml-cpp-dev:
  Installed: (none)
  Candidate: 0.6.2-4+rpi1
  Version table:
     0.6.2-4+rpi1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
```

jwildhome, 2020-06-06:
Good. I will check that out.