GlobalVent/wiki #75: Double check resistance coefficient of valves

carlyjb, 2020-05-15:
Joost to do

Give info to Ralph

jmriphagen, 2020-05-18:
calculated the resistance of exhalation valves  and inspiration valves

ralphBellofattoSTX, 2020-05-18:
can you put the results in this issue?

jmriphagen, 2020-05-21:
[image]
exhaust valve

jmriphagen, 2020-05-21:
[image] this is for the inhalation valve

For inhalation valve 400 mbar -10 peer. dP 390 (going off 10 peep initial pressure )/60.5l/min = 6.55 resistance.

jmriphagen, 2020-05-21:
[image]
D valve

jmriphagen, 2020-05-21:
wall pressure effect on flow

[image]

closed 2020-05-27 by carlyjb