GlobalVent/wiki #95: Demo of GUI with prototype jamvent simulator

carlyjb, 2020-05-22:


ralphBellofattoSTX, 2020-05-24:
system won't build for some reason...

[image]

jwildhome, 2020-05-24:
I was worried that you might have problems building the resource file, Resource.qrc.  You said you had problems with this.  Build the .qrc file just worked for me.

Easiest thing might be to just pull out the *.qrc file for now. Eventually, I assume we will need to get it working.  I was using it to include the fonts in the application, which is not important for now.

Disabling the qrc file should be easy, I can do if we want.  We just need to remove the *.qrc file.  I can do that if we think that makes sense.

$ cat common/resources/CMakeLists.txt
file(GLOB SRCS . *.cpp *.h *.ui *.qrc) **-> file(GLOB SRCS . *.cpp *.h *.ui)** 
set (GUI_SOURCES ${GUI_SOURCES} ${SRCS} PARENT_SCOPE)

ralphBellofattoSTX, 2020-05-24:
I have a feeling there is a file that is not checked into git that is critical to this.

Can you zip up your entire jamventProto directory and post it where i can get to and provide a link.

ralphBellofattoSTX, 2020-05-24:
You had a file PushbuttonWidget.h

But it was referenced as PushButtonWidget.h in the header files.

fixed and pushed fix

ralphBellofattoSTX, 2020-05-24:
[image]

@jwildhome can you get the other graphs going, there are two more dat a streams that can go there.  
Scaling is, well, weird as they are not the "right" data streams"..  but they can act as placeholders

jwildhome, 2020-05-24:
Really good progress.  I will look at this later today or early tomorrow.

jwildhome, 2020-05-25:
Added buttom 2 graphs to show pO2 and lVol.  Not sure of the ranges so
just guessed.  Changed the GUI time out for reading values from 100ms
to 50ms.  Updated GraphWidget::onAddValue() to force value to be within
range.
[image]

closed 2020-05-27 by carlyjb