GlobalVent/wiki #113: Demo I2C detect on raspi

carlyjb, 2020-05-27:


ralphBellofattoSTX, 2020-05-27:
@alcohen 

this should connect an i2c device to the PI's i2c pins,

Follow the directions here:
    * https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c
or here:
* https://radiostud.io/howto-i2c-communication-rpi/

see that the basic i2c detect works.

ajo27, 2020-05-28:
Done - simplest check of i2c sensors being recognized at their correct addresses is successful

Steps:
-Didn't need to install anything (SMbus and i2c-tools already installed and most up-to-date on this Raspbian build)
-sudo raspi-config ---> Interfaces > I2C > Enable
-Connect both pressure sensors
-i2cdetect -y 1

[image]

ralphBellofattoSTX, 2020-05-28:
@alcohen fyi, so you don't end up duplicating the work.  or that you can duplicate the work.

closed 2020-05-29 by carlyjb