GlobalVent/wiki #114: Demo reading i2c sensor on raspi

carlyjb, 2020-05-27:


ralphBellofattoSTX, 2020-05-27:
This demo is linked to this task: https://github.com/JamVent/wiki/issues/113

This takes the demo one step further and demonstrates that we can master programming the i2c interface from c++.

It also proves out any library we want to use to help use this feature.

We should be able to demonstrate that this library also works with a general gpio solution.

solutions that do not require a daemon running are preferable to ones that do.

closed 2020-05-29 by carlyjb

ajo27, 2020-06-01:
Done, using pigpio library. As of 6-1-2020 the simple demo has been taken further to fully match functionality on the Particle Photon - entire MSxxxx library has been ported and we can do "slow" sensor reads with getPressure(), as well as more optimized reads where we do one ADC conversion wait from the main caller function using the broken out functions (sendADCCommand() for both sensor instances inside of main loop, then one delay() call for whichever ADC wait time is greater for the 2 set sensor resolutions, then read back data).

https://github.com/JamVent/prototypes/tree/master/MSxxx_Pi