GlobalVent/wiki #115: select the gpio library we should be using and demo output and interrupt input in c++

carlyjb, 2020-05-27:


ralphBellofattoSTX, 2020-05-27:
we need to be all on the same page about the library to use for gpio and i2c.

NOTE: the most commonly referenced library out there is WiringPI and it is no deprecated and not receiving any updates.  so this is not a choice.

One needs to be selected and demonstrrated that it works for.

* gpio low frequency outputs (valve control)
* gpio inputs -- that can generate a application/user level callback or signal that we can use for our software software design.
* does not interfere with the i2c solution selected.

ajo27, 2020-06-04:
Most basic test in pigpio done -- wire running from 3.3V pin and popping it in/out of GPIO 16 to set if we hit the interrupt handler. Looks good but since this runs in userspace there are potential issues of it missing the event if the thread is asleep when the event happens (see this post for details and workaround using input on 2 gpios to have separate rising/falling edge interrupts, other suggested workaround is DMA which is significantly more involved

) - https://www.raspberrypi.org/forums/viewtopic.php?t=190108#p1196082

[image]

ralphBellofattoSTX, 2020-06-04:
@ajo27 Thanks for the work here.

Ok, so this measured basic connectivity and that means we can get the knob events.  The question is now can we pick them up fast enough.

For the rotary encoder, we only need to trigger on the falling edge, we don't care about the rising for one signal.

The rotary encoder signal when turning it "quickly" by hand seems to be a about a 45 ms pulse.

So the question next will be how quickly is the kernel here waking up the user application to perform the read.

Here is what a hardware debounced signal off "a" rotary encoder looks like.
[image]

I'll have to re-measure when we we are using.  and then come up with an experiment that can at least allow us to estimate the turnaround time from falling edge to getting the callback.

I can definitely see times when we may miss a signal if we spin up too many threads on the rpi that are busy doing stuff and the callback would need to pre-empt a task.

That would be an experiment for another day.  and we we would have to figure out how annoying missing a rotary pulse would be and how often it happens.

closed 2020-06-05 by carlyjb