GlobalVent/wiki #118: GUI screen navigation

carlyjb, 2020-05-27:


carlyjb, 2020-05-29:
Should be done in regards to use cases in dropbox

jwildhome, 2020-06-01:
Will setup this week.  Send email to Alan Walsh on what/who to include in this meeting.

ralphBellofattoSTX, 2020-06-01:
one thing to think about is how to direct the output of the knob and turn it into events that qt can work with.

Right now the interface that is defined is something that can be called periodically and tell you if the button was pressed since the last call and how many clicks left or right the knob has been turned.

It would be possible to add a callback feature to that if that makes it easier to handle.

What probably needs to be done now, is to figure out how to take something like the mouse wheel and button and direct it to something that behaves the same way so you can work out the details of making that work.

I'm not sure if it is possible to direct the mouse wheel events somewhere different than the other focus events are in qt...

jwildhome, 2020-06-01:
From the GUI's perspective, it is easiest to have a callback when knob events are issued.

If we convert the knob events into Qt tab, up arrow, and down arrow key events, we get semi-reasonable behavior almost for free from Qt.

However, it sounds like we will want to do something more complex where we  need to track the elements that can get focus, which element has focus, and the focus state for that element.  All with visuals that make that clear to the user.

closed 2020-07-21 by carlyjb