GlobalVent/wiki #123: Ralph to run proposal for interface by Stephanie & AJ

carlyjb, 2020-05-29:


ralphBellofattoSTX, 2020-05-31:
@ajo27 
@stephmusinsky 

For your review.
proposed classes for hardware with description of how they work.
https://github.com/JamVent/prototypes/blob/master/jamventProto/jamhwlib/JamHw.h
https://github.com/JamVent/prototypes/blob/master/jamventProto/jamhwlib/jamknob.h

alcohen, 2020-06-01:
@jwildhome tagged

stephmusinsky, 2020-06-02:
So youre planning on polling cpld ~ every 10 ms in sync other sensors? 
Might make sense to poll on own slightly more frequently?

What is meant by keepalive? Same as watchdog? But why is there a byte?
Also more info should be sent to cpld such as upon changing valves/states. See: https://www.dropbox.com/s/ep39nrjcmt04evi/CPLD_RPi%20Communications.pptx?dl=0

stephmusinsky, 2020-06-02:
The rotary encoder looks to be a bit different than the primary option I proposed so want to make sure on same page:
You have direction lagging pulse- I was going to have at same time.
Also wasn't going to have an internal counter and just send a pulse each time moves w/ direction. This looks more aggregated of a counted number of pulses and a sole direction?

ralphBellofattoSTX, 2020-06-03:
@stephmusinsky 
10ms interval is for all i2c and serial traffic for the cpld (the linux serial device can buffer a limited amount so at 10 milliseconds we can send a few characters and receive a few) rather than dedicate a thread to it.

This issue on jitter measurement says that we can't reliably poll any faster than 12ms on top of linux.  10ms there is a bit of jitter,  so it won't even be a reliable poll.

The keep alive timeout will have to be much larger than that due to the jitter, so either 100ms time out or 10 10ms keep alive messages before the cpld can consider things dead.     I suspect that watch dog time out is going to eventually have to be about 500ms due to the linux kernel timeslice.

ralphBellofattoSTX, 2020-06-03:
@stephmusinsky 

the rotary interface is just the software class exposing the data.

I was planning that internally the CLK pin would be connected to the system to do a call back on each rising edge.  

When that callback happens then it samples the DT signal and that pin the pi the direction the click is in.  so it should be fine for you to present the signals at the same time as long as you hold it long enough for the linux kernel to wake up the user thread for the event.

The hold time necessary is going to have to be studied.  For this to work we have to be sure that there will always be a core available to process this event, otherwise an event could be missed.

ralphBellofattoSTX, 2020-06-03:
ok, after actually looking at the raw output of a rotary swtich, and the fact that a phase I simulation test board is going to have to be debounced with a cap.  it looks like it would be best if i worked off the falling edge of the encoder rather than the rising edge, it is a lot crisper and we get a clean signal out of the DT at that point too.

@stephmusinsky So, to you since this is coming from the cpld, if you truely can present both signals at exactly the same time, then this should work.  

I'll have to measure the time that it takes to read the data from the rpi for us to decide if we have any issues with hold times.

were you planning to present both signals whenever there was any transition, or only the clk signal transitioning?

ralphBellofattoSTX, 2020-06-03:
Without debounce we see:
[image]

hardware debounced signal
[image]

[image]

So with a real encoder with 100 nf we get a nice clean falling edge:
[image]

stephmusinsky, 2020-06-04:
I was planning on sending pulse and direction for every knob turn. Programmable hardware is great for timing things at same time.

ralphBellofattoSTX, 2020-06-04:
I believe that will work, we just have  to make sure the hold time is enough for the hold time on the dt is long enough such that we don't miss it.

I'll have to do a measurement to determine how long that is to determine the latency time from the falling edge to the callback.

I have an experiment in mind, where i can measure with a scope the time from the falling edge of the input signal to the time it takes the rpi to wake up the thread waiting on the IO and then perform a gpio write.  That will also measure the time it takes to do a write io, so it that measurement will be longer than just the callback time, but should give us a number that will tell us what our design point should be.

stephmusinsky, 2020-06-04:
Yeah, I’ve asked for a time length to make sure pulse lasts for. Let me know when you have a good estimate

ralphBellofattoSTX, 2020-06-04:
I'll need to measure with the devices we have on our board, but a measurement with one i have on my shelf, is about 45ms, when turning the knob "fast" by hand,  I could get a higher pulse rate and shorter duration if i put it in a controlled speed source or a hand drill.  

While that would be an interesting exercise, i'm not sure of the value for this project.

carlyjb, 2020-08-11:
Tickets #83, #214, #215, and #216 should capture implementing this interface.

closed 2020-08-11 by carlyjb