GlobalVent/wiki #126: Thermal run with temperatures of sensitive components

dcstraney, 2020-05-29:
Test conditions:
* Measure ambient temp. outside case, to calculate temperatures at worst-case ambient
* High thermal load: valves cycling at high but realistic duty cycle, with PWM; Pi running some reasonably intensive compute load; possibly electronics high-current fault? (one power supply shorted)

Temperatures to measure:
* Valves?
* Pi's processor
* PCB's input Schottky diode
* PCB's power supplies

Possible mitigations:
* Bigger fans
* Direct contact of valves against metal mounting plate (with gap pad, etc. in between)
* Direct contact of PCB against metal mounting plate (with gap pad, etc. in between)
* Heatsink and/or extra fan on Pi

dcstraney, 2020-05-29:
Waiting on integration with vaguely-representative electronics, mechanical assembly, and fans

ralphBellofattoSTX, 2020-05-29:
static measurements i have a 4 channel reader and thermal couples,

A data logger has a 2channel amp, but requires some software to either be written or located that converts the thermal couple voltage to a temperature.

It has been a while since i have used the data logger for thermal measurements.