GlobalVent/wiki #129: proposal for integration testing for cpld and rpi.

ralphBellofattoSTX, 2020-05-31:
We need a way to do integration testing with the following:
* cpld
* raspberry pi
* hardware board with valve drivers and pressure sensors.

Once we do the second run of hardware boards, there will be more of those boards available than there are jamvent assemblies with gas supplies.

We need a way to do integration testing.

I propose we hook up the following:

Rpi ----> hardware board with cpld --> photon

Where the photon is connected to the same i2c bus chains that the cpld and the rpi are connected.

This would require the photon to contain the following:

* the software jamvent simulator (already written).
* 2 I2c slave asynchronous state machines.
* a software model of 2 sets of:
  * High pressure sensor.
  * low pressure sensor.
  * A/D reader for the O2 sensor.

The pressure sensor model would have to reverse all the conversion polynomials.  However, initially we could cheat, and when in the test configuration the reading is the actual pressure with no conversion.

In addition to the emulation of the I2c devices, both the rpi and the cpld should have test configurations that would allow it to use 3 alternate i2c addresses that the photon will respond to on the same i2c busses.

in addition this configuration can also be setup to test the pressure and o2 sensor reading by connecting them up to pneumatic test set up.

For example, a soda bottle, fizz keeper pump, a couple of 3d printed parts and aquarium tubing can be used to test the pressure sensor reading by both the cpld AND the rpi, and make sure we don't have any noise problems with the measurements.  Low pressure readings can also be verified with a home made manometer.  Low pressure readings would require an open ended one, and higher pressure readings a closed ended one.

Then a test configuration can be switched to and the entire control alogrithms on both the cpld and the rpi can be tested out.

@stephmusinsky 
@ajo27 

Idea for your review.  Please  include your comments in this issue.

ralphBellofattoSTX, 2020-06-02:
@alcohen @ajo27 @stephmusinsky @jwildhome 

block diagrams as requested.
[image]

[image]

[image]
[image]

Inside the photon would live the existing state driven model and 2 pairs of i2c slave recievers

The I2c slave receivers would be a 2 level state machine.
* bit level (identify, start/stop/bitshift)
* byte level (start, start, byte read, byte write events).

C++ connected to the I2c slave state objects emulate each of the i2c sensors and converters.

I2c specification is spelled out here.
https://www.nxp.com/docs/en/user-guide/UM10204.pdf

Bonus points will be given for implementing a 4th i2c device which will allow writes to control registers to change parameters of the underlying simulation

someones attempt at an i2c state diagram.
[image]

ralphBellofattoSTX, 2020-06-02:
Photon software structure for this:
[image]

stephmusinsky, 2020-06-02:
Thanks for making these!
 Phase II- CPLD should be able to read rpi valve signals (and maybe setup "AND" to LEDs like Don's actual board?). Same for phase III
Why does CPLD not have I2C in phase III? Ohh is it just inside that block as in the production board

Also just a note so dont forget but would be nice for CPLD to have a low and high priority led alarm indication

ralphBellofattoSTX, 2020-06-03:
@stephmusinsky 
Phase III diagram is incorrect, i'll route the second i2c set to the cpld.

i can add a couple of generic leds for the cpld to light up if you want.

ralphBellofattoSTX, 2020-06-03:
@stephmusinsky I revised the pictures here:
https://github.com/GlobalVent/wiki/issues/129#issuecomment-637539347. adding in the leds you requested.

ralphBellofattoSTX, 2020-06-03:
NOTE: 100nf debounce caps seem to give us the best result, with 1nf we noise got through.

Very slow rise time, but it gets back quick enough for the next click...

stephmusinsky, 2020-06-04:
Thanks for update. For phase II, I still think the rpi valve signals should be routed to cpld as well so it can read. Also not sure if we want an “and” Setup like dons board has to the valve leds. So both cpld and rpi have to turn on to be on. Cpld will be defaulted to all on unless an error occurs that it should take over

ralphBellofattoSTX, 2020-06-04:
right you want to see the valve signals as well...  I'll revise.

ralphBellofattoSTX, 2020-06-04:
I'll be picking out pins on the photon side today and note that in the pictures as well.

stephmusinsky, 2020-06-04:
It’s probably easiest for don if you use the same pins he designated in pinout for other boards

ralphBellofattoSTX, 2020-06-04:
same pins on the rpi side, but, i need to pick the photon pins...

ralphBellofattoSTX, 2020-06-04:
proposed pinouts and what i'm going to wire on a breadboard, right now.

[image]

ralphBellofattoSTX, 2020-06-05:
phase I bread version ready to test:
[image]

closed 2020-06-08 by carlyjb