GlobalVent/wiki #131: Set goals for next prototype/demonstration/integration test

alcohen, 2020-06-01:


alcohen, 2020-06-03:
I propose that for our next integration/demonstration milestone we have our hardware and software support mandatory breathing. This would be with a UI that's limited to showing only what's reasonably necessary for this task. Perhaps on next call we can discuss if this seems like a reasonable approach; if it is, the we can work out what this means from clinical and technical standpoints, then determine what more must be done for that the happen, and timing. This might have the added benefit of giving us something that we can bring into the MGH respiratory lab for preliminary testing.

ralphBellofattoSTX, 2020-06-03:
From where I sit we need the following software implemented to do the "mandatory breathing"
* gas mix control software
* control software that actually controls the breathing cycle.
* rpi software that actually talks to hardware.
* software to deal with the rotary knob
* connection of rotary knob software to ui

Software needed to enable the above:
* to support the last element, we are going to need the test Jig i outlined in:
https://github.com/GlobalVent/wiki/issues/129
  * Software on the photon required.
    * i2c slave.
    * emulation of the pressure and a2d i2c devices.
    * integration of the above with the software jamvent emulation.

And, anything else i may have missed.

carlyjb, 2020-06-15:
Captured in https://github.com/GlobalVent/wiki/milestone/4
Can be moved to a future milestone if determined they need more time.

closed 2020-06-15 by carlyjb