GlobalVent/wiki #158: Proposal API for GUI/Controller Interface

carlyjb, 2020-06-08:
Make the interface serial/i2c agnostic for the GUI?

Alan to write out an API to propose to @jwildhome and @ralphBellofattoSTX 

Part of a question about is it worth using the Photon as a quick patch.

ralphBellofattoSTX, 2020-06-09:
This has already been done and an abstract base class has already been created to do this.

Please refer to the interface defined here:
https://github.com/GlobalVent/prototypes/tree/master/jamventProto/jamhwlib

To use it, it needs to be filled out and then connected to a control library:
https://github.com/GlobalVent/prototypes/tree/master/jamventProto/jamctrllib

and as for an overall design that also calls out for this abstraction is also articulated here.
https://github.com/GlobalVent/wiki/issues/31#issuecomment-616689746

So, before people ask for more work on this, please review the information already provided.

alcohen, 2020-06-10:
Ralph, I don't think that is this? We were discussing the specification for data/commands going back and forth over serial, e.g., pressure coming as a stream of numbers with a letter at the start of each number that identifies which sensor. For example, H123 L234 H124 L232...

jwildhome, 2020-06-10:
I am in the process of restructuring the GUI so that the data communication is separated from a specific screen.  For convenience now, we have the JamCtrlMgr connected to the Treatment screen.  I want to uncouple to make the communication available to other screens as I add them.

I got a little over my head doing this all at once, so am breaking down into smaller steps.
1. Move use of JamCtrlMgr out of main.cpp to a CommMgr instance.  This step is mostly done.
2. Make CommMgr data available to screens.
3. Add the Powerup screen.
4. Make the GUI communication interface general enough to work with the JamCtrlMgr and the serial interface.

I would like to point out that the "shape" of the 2 interfaces is different and I am trying to see if I can abstract that away.  JamCtrlMgr using polling, and the serial interface is driven as the data arrives.

closed 2020-06-12 by alcohen