GlobalVent/wiki #166: Investigate integrated CPU options

dcstraney, 2020-06-12:
Just some thoughts on possibly replacing the Pi with a single CPU chip at some point in the future (the STM32 series has been suggested as a candidate):
+ More integrated, less mechanical assembly steps and easier sourcing (as long as chip is highly available)
+ More compact, more mechanically and thermally robust (with lower power)
+ Flash memory is likely more long-term reliable than SD card
+ More reliable loop timing with no OS overhead
- Display is harder; seem to be some displays with cross-series-compatible interfaces available (~2K avail. total) but still not as universal as using any monitor with HDMI port; HDMI transmitter requires high-speed signal verification, becoming member of HDMI association
- Networking (if added for remote monitoring, etc.) would be doable but harder

ralphBellofattoSTX, 2020-06-12:
not so sure about the "no os overhead" just the exploration with the photon is showing os overhead is there as well, it is not exactly bare metal unless you want to write all the drivers for the hardware interfaces oneself...

dcstraney, 2020-06-14:
"no overhead" doesn't mean there's going to be no delays in anything, but that there's not going to be tons of background processes as in Linux or another "full" OS; if we used an RTOS (like FreeRTOS) or a set of pre-written drivers with no OS/threading at all then the timing would be much more predictable.  How much it actually matters I don't know - but that was my point on that item.

ralphBellofattoSTX, 2020-06-14:
ok, well as we have seen with the photon, which i believe is running an RTOS, there is still a fair amount of unexpected overhead.

Bare metal with drivers to do only what we need is probably what is in order.  Finding a board with a dev environment similar to what is available for the photon (as a VScode extension) is pretty much a requirement for any dedicated device.

the photon may be more compatible with. some of this if one could spend the time turning off background processes that are not needed.

I think the study done in preparation for doing the i2c slave, shows us the kind of things needed to verify prior to committing to a design.

ralphBellofattoSTX, 2020-06-14:
I suspect that the photon has already grafted a lot of its wifi stuff on top of freertos, so it is not quite as bare metal as one would like.

Also, some of the functions that the low level code are handling (such as reversing direction of a line), seem to take way longer than I would have expected..    So I think the lesion here is, after reading the specification, measure...

Also the amount of background jitter also seems to be a bit surprising.