GlobalVent/wiki #168: Website: Short Bios (& Photo)

alcohen, 2020-06-14:
Each team member adds a super-short bio and and adds a photo to Fundraising\Website.

Bio format should be similar to this:

_Don Straney is an electrical engineer with about a decade of experience, who enjoys and has gravitated to challenging problems ranging from precision analog measurement circuitry to high-speed signal processing and cost-constrained high-frequency power conversion.  He currently works for the A.A. Martinos Center for Biomedical Imaging at Mass General Hospital, building custom research equipment"

We may not use photos at first if they are inconsistent in the way they look.

carlyjb, 2020-06-15:
- [ ] Alan C
- [ ] Alan W
- [ ] Joe W
- [ ] Carly B
- [ ] AJ O
- [ ] Ralph B
- [ ] Bryan G
- [ ] Don S
- [ ] Steph M
- [ ] Joost R

carlyjb, 2020-06-15:
Located in Fundraising/WEbsite

bglezerson, 2020-06-17:
I need to find (take?) a picture of myself to upload. Have uploaded a bio, however.

closed 2020-06-29 by carlyjb