GlobalVent/wiki #170: Make GUI structure configurable

carlyjb, 2020-06-15:
Run time? Compile time? (preference for run)

jwildhome, 2020-06-29:
I am running into problems trying to get this to work with YAML.  I was able to install and configure the yaml-cpp library on the RPI and set up the VisualGDB environment to use.

However, the CMake find_package(yaml-cpp) is failing and I cannot figure out why.  It all looks correct.  I tried using a manual alternative to find_package(), which builds but I had to use a file path from my setup and have not figured out how I can do in a general way.

At this point, I spend 6 or more hours trying to make it work.  I think I will just drop back to using JSON which is available in the Qt Core library and I am more familiar with.

jwildhome, 2020-07-04:
I added a Config.json file for configuration.  As mentioned above, I switched from YAML to JSON as I was unable to the YAML library to build in this environment.  I think JSON is good enough for now.  Attached is the Config.json file.  We can add to as we see fit.

P.S. Had to change suffix to .txt to attach here.

I consider this task completed and will close.

[Config.json.txt](https://github.com/GlobalVent/wiki/files/4872901/Config.json.txt)

closed 2020-07-04 by jwildhome