GlobalVent/wiki #182: Integrate encoder

carlyjb, 2020-06-29:


carlyjb, 2020-07-06:
Fighting a bit with QT events.
Focus on functional first. Eventually will need to implement a better solution.
Later: demonstrate showing focus changing.

jwildhome, 2020-07-08:
Updated GUI so can navigate GUI with tab, up arrow, and down arrow keys.

So can inject knob events as these key events and get a 1st pass at functionality.

This is sort of a poor man's navigation that basically comes for free from Qt.  Knob press mapped to tab key event. CW (clockwise) click mapped to up arrow key. CCW (counter clockwise) click mapped to down arrow.

So a knob press moves focus to the next focusable element (spin box or button).  CW/CCW click increments/decrements spin box (like FiO2) value for spin box elements.  CW/CCW click activates the button for button elements.

I'm still thinking through the full functionality for navigation with knob events.